Xilinx/Vitis-Tutorials #143: ERROR: [v++ 60-602]

canozcifci, 2021-12-17:
Hi,

I am facing a problem when i build vector addition application project.  I can build with Emulation-SW, but when I tried to build with Emulation-HW or Hardware. I got the error:
ERROR: [v\+\+ 60-602] Source file does not exist: /Xilinx/kv260-vitis/platforms/ws/add_vector_00_system_hw_link/Emulation-HW/binary_container_1.xclbin
INFO: [v\+\+ 60-1662] Stopping dispatch session having empty uuid.
INFO: [v\+\+ 60-1653] Closing dispatch client.
makefile:38: recipe for target 'package' failed
make: *** [package] Error 1.

Thanks
John

imrickysu, 2021-12-17:
Hi John,

binary_container_1.xclbin is the output of v++ link. I guess your v++ link doesn't complete successfully. Could you check whether there are any error information?

canozcifci, 2021-12-21:
Hi Ricky, 

Thanks for the quick reply. I followed the instructions "Vitis 2021.1 Embedded Platform for Zybo-Z7-20" 

[](https://highlevel-synthesis.com/2021/08/15/vitis-2020-1-embedded-platform-for-zybo-z7-20/)

Can you tell me where i can find the necessary error info for v++ link?

Thanks
John

imrickysu, 2022-01-05:
Hi @canozcifci , if you select the hw_link component under the system project, it will show you related building info. 

When selecting the system project, it only shows you the packaging related info.

Seyyidosman, 2022-03-14:
Hi, I am also facing this problem. Please help me!

Seyyid

mor2810, 2023-03-01:
any fix for that issue ?
i have exactly the same error

AnusheelXilinx, 2024-12-23:
If there is an error in the Vitis build run, user may encounter this issue. Please check the log files to understand the failure. 
Please test the latest tutorial steps and let us know if you still this issue while building the design. 

We do not see this issue in the internal testing. 

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx