Xilinx/Vitis-Tutorials #145: The Getting-Started app.exe segfaults on XRT 2.13.0

vkomenda, 2021-12-22:
I tried the getting started example on my Varium C1100 aka u55n. I have neither u200 nor zcu102 to compare with. The XRT version is 2.13.0 from the Git master branch.

Both `sw_emu` and `hw_emu` end with the same errors in `app.exe`. Here is the listing from a `hw_emu` build:

```
...
cp xrt.ini hw_emu
cd hw_emu && XCL_EMULATION_MODE=hw_emu ./app.exe
XRT build version: 2.13.0
Build hash: 37f368eae34cb70b8c3dc1c840d02e59b2b6b8cb
Build date: 2021-12-22 13:23:51
Git branch: fix-ref-error-XclBinClassCxx
PID: 11784
UID: 1000
[Wed Dec 22 17:32:51 2021 GMT]
HOST: noether
EXE: /home/vk/src/x/Vitis-Tutorials/Getting_Started/Vitis/example/u55n/hw_emu/app.exe
[XRT] WARNING: The xrt.ini flag "opencl_summary" is deprecated and will be removed in a future release.  A summary file is generated when when any profiling is enabled, so please use the appropriate settings from "opencl_trace=true", "device_counter=true", and "device_trace=true."
INFO: Found Xilinx Platform
INFO: Loading 'vadd.xclbin'
[XRT] ERROR: Enable multithreading to use std::thread: Operation not permitted
[XRT] ERROR: Cannot allocate buffer at unknown memory index
[XRT] ERROR: Cannot allocate buffer at unknown memory index
[XRT] ERROR: Cannot allocate buffer at unknown memory index
```

Is this a known issue? How can I debug this?

randyh62, 2021-12-30:
It could be a problem with your setup. Can you provide some additional details with regard to your OS? 

I just ran hw_emu on the xilinx_u55n_gen3x4_xdma_2_202110_1 and it built an ran fine. I did have to eliminate the u200.cfg to make it build properly, but that did not cause any issues. The XRT version info is:
XRT Build Version: 2.12.456 (2021.2)
Build Date: 2021-12-14 23:06:37

You can download the setup files for that card from this location: 
https://www.xilinx.com/products/accelerators/varium/c1100.html#gettingstarted

You could also try to build and run it on the U200 to see if that will run with your current XRT installation.

vkomenda, 2021-12-30:
@randyh62 Here is the output of `xbutil examine` with some OS details. It also shows the problem I'm facing. I described it in my [support forum post](https://support.xilinx.com/s/question/0D52E00006uVLA2SAO/u55n-flashing-the-shell-partition-doesnt-work?language=en_US&t=1640880209936)

```
$ sudo /opt/xilinx/xrt/bin/xbutil examine --report all
Device list
  [0000:61:00.1] : xilinx_u55n_gen3x4_xdma_base_1 
Warning: Due to missing device, the following reports will not be generated:
         - aie
         - aieshim
         - memory
         - dynamic-regions
         - debug-ip-status
         - error
         - pcie-info
         - platform
         - electrical
         - mailbox
         - mechanical
         - firewall
         - thermal
         - qspi-status
         - cmc
System Configuration
  OS Name              : Linux
  Release              : 5.15.0-2-amd64
  Version              : #1 SMP Debian 5.15.5-2 (2021-12-18)
  Machine              : x86_64
  CPU Cores            : 32
  Memory               : 257639 MB
  Distribution         : Debian GNU/Linux bookworm/sid
  GLIBC                : 2.33
  Model                : System Product Name

XRT
  Version              : 2.13.0
  Branch               : fix-ref-error-XclBinClassCxx
  Hash                 : 37f368eae34cb70b8c3dc1c840d02e59b2b6b8cb
  Hash Date            : 2021-12-22 13:23:51
  XOCL                 : 2.13.0, 37f368eae34cb70b8c3dc1c840d02e59b2b6b8cb
  XCLMGMT              : 2.13.0, 37f368eae34cb70b8c3dc1c840d02e59b2b6b8cb

Devices present
  [0000:61:00.1] : xilinx_u55n_gen3x4_xdma_base_1 user(inst=129) 
```

vkomenda, 2021-12-30:
@randyh62 Can you please give me a download link for xrt 2.12.456 as in your system. Unfortunately xrt 2.11.634 fails to build the DKMS drivers on Ubuntu 20.04, kernel 5.11.

vkomenda, 2021-12-30:
I downgraded the kernel to 5.8.0-63 and that allowed the 2.11.634 DKMS drivers to be built. Also the `sw_emu` example target built OK.

The `hw_emu` target fails with
```
terminate called after throwing an instance of '__gnu_cxx::recursive_init_error'
  what(): std::exception
Aborted (core dumped)
```

I wonder how much version-matching will be required just to get this hello world up and running. I can't use kernels that are old. My hardware is new. My use case requires newer kernels with fixed CVEs.

vkomenda, 2021-12-30:
Oh, even though xrt built the DKMS modules, the `xclmgmt` module is useless.
```
modprobe: ERROR: could not insert 'xclmgmt': Unknown symbol in module, or unknown parameter (see dmesg)
```

and in `dmesg` the missing symbols are
```
fpga_mgr_create
fpga_mgr_unregister
fpga_mgr_register
fpga_mgr_free
```

So again the version-matching game is lost on my end.

imrickysu, 2022-01-05:
Hi @vkomenda , XRT 2021.2 was tested with kernel 5.4 on Ubuntu 20.04 because HWE is not supported.
Please refer to the tested version details in answer record https://support.xilinx.com/s/article/XRT-Xilinx-Runtime-version-2021-2-2-12-Tested-Operating-Systems?language=en_US and XRT release notes https://www.xilinx.com/content/dam/xilinx/support/documentation/sw_manuals/xilinx2021_2/ug1451-xrt-release-notes.pdf

keryell, 2022-01-05:
Yes, it would be nice to have support for modern software too.

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx