Xilinx/Vitis-Tutorials #148: Petalinux (2021.2) fails to build for Vitis Accel flow

Akteevy, 2022-01-03:
I followed the Vitis platform creation tutorials (parts 1 and 2) here:
https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step1.md

When running petalinux-build, this error happens:
`| ext_platform:
| Failed to find amba_pl node !!!
| ERROR: [Hsi 55-1545] Problem running tcl command ::sw_device_tree::generate : Failed to find amba_pl node !!!
|     while executing
| "error "Failed to find $lu_node node !!!""
|     (procedure "get_node_object" line 31)
|     invoked from within
| "get_node_object amba_pl pl.dtsi"
|     invoked from within
| "create_dt_node  -name zyxclmm_drm -objects [get_node_object amba_pl pl.dtsi]"
|     ("eval" body line 1)
|     invoked from within
| "eval "create_dt_node ${cmd}""
|     (procedure "add_or_get_dt_node" line 187)
|     invoked from within
| "add_or_get_dt_node -n "zyxclmm_drm" -d ${default_dts} -p $bus_node"
|     (procedure "gen_zocl_node" line 21)
|     invoked from within
| "gen_zocl_node"
|     (procedure "::sw_device_tree::generate" line 23)
|     invoked from within
| "::sw_device_tree::generate device_tree"
| ERROR: [Hsi 55-1442] Error(s) while running TCL procedure generate()
| hsi::generate_target: Time (s): cpu = 00:00:04 ; elapsed = 00:00:05 . Memory (MB): peak = 2306.863 ; gain = 0.000 ; free physical = 32386 ; free virtual = 47131
| generate_target failed
|     while executing
| "error "generate_target failed""
|     invoked from within
| "if {[catch {hsi generate_target -dir $project} res]} {
| 	error "generate_target failed"
| }"
|     (file "/home/joan/workspace/Petalinux/zcu104_custom_plnx/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/dtgen..." line 49)
| WARNING: /home/joan/workspace/Petalinux/zcu104_custom_plnx/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.62513:176 exit 1 from 'eval xsct -sdx -nodisp /home/joan/workspace/Petalinux/zcu104_custom_plnx/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/dtgen.tcl -ws /home/joan/workspace/Petalinux/zcu104_custom_plnx/project-spec/configs/../../components/plnx_workspace/device-tree -pname device-tree -rp /home/joan/workspace/Petalinux/zcu104_custom_plnx/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/git -processor_ip psu_cortexa53 -hdf /home/joan/workspace/Petalinux/zcu104_custom_plnx/build/tmp/deploy/images/zynqmp-generic/Xilinx-zynqmp-generic.xsa -arch 64 ${APP_ARG} ${MISC_ARG}'
| WARNING: Backtrace (BB generated script):
| 	#1: do_configure, /home/joan/workspace/Petalinux/zcu104_custom_plnx/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.62513, line 176
| 	#2: main, /home/joan/workspace/Petalinux/zcu104_custom_plnx/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.62513, line 210
| 
| Backtrace (metadata-relative locations):
| 	#1: do_configure, /home/joan/workspace/Petalinux/zcu104_custom_plnx/components/yocto/layers/meta-xilinx-tools/classes/xsctbase.bbclass, line 54
ERROR: Task (/home/joan/workspace/Petalinux/zcu104_custom_plnx/components/yocto/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7624 tasks of which 6417 didn't need to be rerun and 1 failed.
`
This seems to happen for any platform configured as Vitis extensible (I have no problems for "normal" embedded flow platforms).
I tried a number of variations in the configuration, but always get exactly the same. Even for a custom generic zynqmp machine.

imrickysu, 2022-01-04:
Hi @Akteevy, could you try to use the Makefile generated XSA? 

```bash
cd Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/ref_files/
make step1 step2
```

Hi @sven2314 , could you help to investigate and follow up?

Akteevy, 2022-01-04:
I tried the above and failed, but... at fetching with git. See log attached.
Then I went into the petalinux folder inside step2, ran petalinux-build (previously changed it to use cache downloads) and completed without errors.
So there is something different in my configuration, even if I looked at it a number of times.
[build.log](https://github.com/Xilinx/Vitis-Tutorials/files/7809948/build.log)

Akteevy, 2022-01-04:
I opened the vivado project created by make as above and noticed the interrupt controller was implemented while I didn't have it (apparently is optional). I rebuilt my project with the interrupt controller but got exactly the same error, so another suspect cleared.
I also had the suspicion of the package set, as I had to fiddle a bit to install libsdl2-dev (downgrading some other packages), but the fact that make builds it makes me think that's not a problem.

imrickysu, 2022-01-05:
Hi @Akteevy , your build.log says fetch failed because "github.com (port 9418) (Temporary failure in name resolution)". Please check your network settings. 

To quickly test whether your XSA caused the device tree generation error, you can use the script here: https://github.com/Xilinx/Vitis_Embedded_Platform_Source/blob/2021.2/Xilinx_Official_Platforms/xilinx_zcu102_base/sw/prebuilt_linux/Makefile#L30 
It uses the standalone device-tree-generator. PetaLinux is not needed.

sven2314, 2022-01-05:
@Akteevy
I tested your trial with setting it as Vitis extensible platform. when building PetaLinux I did not encounter your error. I want know your ENV. Can you tell me the version you are using of Vivado , also Vitis ? If possible can you share me your Vivado project. I want to reproduce your phenomenon.

Akteevy, 2022-01-05:
Hi @sven2314 , I'm using 2021.2 release installed on a Ubuntu 20.04.1.
Attached is the TCL for the vivado project

[project.tar.gz](https://github.com/Xilinx/Vitis-Tutorials/files/7817103/project.tar.gz)

sven2314, 2022-01-06:
Hi, @Akteevy 
I tried your project with error that could not find board definition. I checked the TCL file and found you were using board from Myir. As I do not have this board file in my Vivado. So in order to go on to find the root cause please share your board file for Vivado.

Akteevy, 2022-01-06:
Hi @sven2314 
That's true, attached are the boards files, to be untared to <Vivado install folder>/2021.2/data/boards/board_files/

[ultramyir.tar.gz](https://github.com/Xilinx/Vitis-Tutorials/files/7820509/ultramyir.tar.gz)

Akteevy, 2022-01-06:
I tried applying the tutorial makefile (with the only difference to use the cached sstate and downloads) to a project built for the zcu104 with the same error, so I guess the problem is in the Vivado side/ XSA

sven2314, 2022-01-08:
Hi, @Akteevy 
I tried your project with Myir board file added. After running one time automation  I exported the platform with two methods. One was GUI and other one was command line. Both were ok for PetaLinux compilation.  If you need I can share my project ZIP file. Or if you are convenient please share your project ZIP file. So that I can open the project directly and  help to check the differences between mine and yours.

Akteevy, 2022-01-10:
Unfortunately and because of lack of space in the VM I deleted the project that built. I've been trying to replicate it but didn't succeed. I don't think now that exporting with the command makes a difference. I also tried building the XSA in windows but got the same result. I have been using the cached sstate, I will next try the slow way.

Akteevy, 2022-01-13:
After running this script (https://support.xilinx.com/s/article/73296?language=en_US) to check the installed packages it did some changes on them and I was able to build the example for zcu104 board with make step1 step2 (even if I modified them to use the cache sstate). But still I can't build a similar project for this MYIR board (MYD-XCZU3EG). I even tried changing the makefiles and tcl to build to my specific board.
I've been comparing the build folder for both projects (the now building zcu104 example and the for the myir board) and one difference I noticed is the absence of dts and dtsi files in <project>/components/plnx_workspace/device-tree/device-tree
where the building project has a system-top.dts and the failing one doesn't.
[image]

Akteevy, 2022-01-16:
I could finally go past that error by selecting 'Remove PL from device tree' in the general config -> DTG Settings. At least I can create the Vitis platform and compile an example. Not booting yet as I have some problem with the rootfs location, but doesn't seem a deal breaker.

sven2314, 2022-01-19:
@Akteevy Hi, I checked the project directory which use your board file. There is system-top.dts.
By the way Could you share the name you set for the machine when you set the machine name in Step2?

Akteevy, 2022-01-19:
I left the board name as 'template' being a non-supported board (I think it just fails and pops that as an error if using any board name not in the list in the petalinux manual).
Now I just checked there is a system-top.dts generated (in the folder above) **for the case where I excluded the PL** from the device tree in the configuration. In that case, petalinux and Vitis build without problem.

Akteevy, 2022-02-03:
I could complete building and running a Vitis accelerated application, and I think it is necessary the above (selecting the 'remove PL from device tree') in the dTG settings, now creating a detailed tutorial that I'll publish.