Xilinx/Vitis-Tutorials #149: Could not find the input file when running the program during hardware emulation

Candice-Yin, 2022-01-07:
I used absolute path when I created my host program as show below:
[image]
But when running the program using "./app.exe" during hw_emu, it displayed as show below:
[image]

Does anyone know why?
Thank you!

imrickysu, 2022-01-07:
Hi @Candice-Yin , please provide evidences that these files exist in your emulation environment.

Please note that these files should be in your rootfs, not on your x86 host machine because hardware emulation cannot access x86 filesystem.

It's better to add your files in sd_dir so that these files can be packaged into FAT32 partition of sd_card.img. These data files will be in the same directory as your app.exe in /media/sd-mmcblk0p1.

For more info about sd_dir, please refer to https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step3.md

Candice-Yin, 2022-01-07:
Thank you @imrickysu I solved it. 
I‘m not sure about one thing: how can I connect the data files that host progam reads with the input data that the source code for the kernel needs? By the same  name of variables?

imrickysu, 2022-01-10:
Sure. I'm closing this issue.

For the question about feeding data to kernel, please read docs about your kernel API. Data in user space and kernel space are separated. It needs one data copy to send data from user space to kernel space, or use kernel API to allocate memory space and fill in data to the data pointer.

closed 2022-01-10 by imrickysu