Xilinx/Vitis-Tutorials #152: Error when running Hardware_Acceleration/Introduction/guided_sw_examples tutorial

Capser-Minyong, 2022-01-11:
Hi, I got error when running guided_sw_examples tutorial using U250 on CentOS7 server. 

[image]

From the log, I found that m_axi port bundle name doesn't make between connectivity_u250.ini and wide_vadd.cpp

[image]

[image]

(Also only 1 sp for vadd.cpp is describe on connectivity_u250.ini. shoud I insert additional two sp for vadd.cpp ? )

[image]

After I modified connectivity_u250.ini, I passed error above but still got error for vpl. 

[image]

Do you guys have any idea what is wrong with my setting for tutorial?

Thanks,
Casper

Capser-Minyong, 2022-01-11:
And I also have question for using vitis vision library.

[image]

 First I got errors for missing header (common/xf_common.hpp), I couldn't find way to import this header, I git cloned Vitis_Libraries(https://github.com/Xilinx/Vitis_Libraries) on hw_src folder.
After that, I passed error above but I am not sure it is right way to solve error.

As many examples on Hardware_Acceleration/Design_Tutorials need Vitis_Libaray, locating Vitis_Libararies on all tutorial folder is impossible. Is there any proper ways to use Vitis_Libararies among tutorials?

Thanks,
Casper

allyzhou, 2022-01-19:
@randyh62 can you take a look at this issue, Thanks.

randyh62, 2022-01-21:
Hi Casper, The tutorials uses the Vitis_Libraries/vision/L1/include as indicated in the design_sources/hw_src/Makefile:
`VISION_LIB_FLAGS := -I. -I./Vitis_Libraries/vision/L1/include --config vision_config.ini`

The installation and setup instructions are kind of vague (missing) on this point, but you can find the required file here:
[https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L1/include/common](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L1/include/common)

As to your VPL error, it could be a memory issue? Not sure what your system configuration looks like, but I don't see any other indication. I would suggest run it again and if you still have a problem, look for the Vivado log file which you should be able to find at: `hw_src/_x/link/vivado/vpl`

Also, can you post your modified `connectivity_u250.ini` so I can see what changes were needed to run the tutorial? 

Best regards,
Randy

Capser-Minyong, 2022-01-21:
Hi, randyh62. Thank you for fast reply.

> You can modify the Makefile in the hw_src folder to point to your Vitis_Libraries installation or the path to it.
Modify Vitis_Libraries path is a good solution for further tutorial examples.

> I would suggest run it again and if you still have a problem
I did make clean and ran it again. I finished it successfully without any error after that.

> Also, can you post your modified connectivity_u250.ini so I can see what changes were needed to run the tutorial?

below is my connectivity_u250.ini (which I finished build sucessfully)

[image]

I modified 2 parts.

1. I modified sp of wide_vadd bundle name according to wide_vadd.cpp

wide_vadd.cpp
[image]

2. I modified vadd.cpp and sp of vadd bundle name according to vadd.cpp

vadd.cpp baseline
[image]

I modified vadd.cpp Interface declear similar to wide_vadd.cpp
[image]

randyh62, 2022-01-24:
Thanks Casper. I am going to update the ini files as you have suggested and close this issue. 

Regards,
Randy

closed 2022-01-24 by randyh62