Xilinx/Vitis-Tutorials #153: Does Vitis support VCK5000?

whyaysd, 2022-01-11:
Does Vitis support VCK5000 platform? 

Could you please give me an example project targeting vck5000?

Premduth, 2022-01-11:
@allyzhou  can you comment?

whyaysd, 2022-01-13:
@Premduth  @allyzhou  Could you please give me some advice?

allyzhou, 2022-01-13:
@whyaysd depends on different versions of VCK5000 boards and platforms, Vitis provide a different level of support. If you have VCK5000 for Vitis AI, the demo are publically accessible at https://github.com/Xilinx/Vitis-AI. Other than Vitis AI on VCK5000, we don't have public accessible tutorials/examples. We do have a plan to release a Vitis tutorial on VCK5000, but it will only be an EA version through a secure site like a lounge.

whyaysd, 2022-01-13:
Thanks for your information.

I am participating in the Adaptive Computing Challenge 2021 hosted by Xilinx: https://www.hackster.io/contests/xilinxadaptivecomputing2021. I need to use VCK5000 to finish my work, and I need to use Vitis to implement my functions. 

I have got the EA version of VCK5000 from xilinx and I have access to the secure site: https://www.xilinx.com/member/vck5000-aie.html#overview.

Taking those into consideration, could you please help me to solve those problems or just give me a demo?

Thank you

whyaysd, 2022-01-17:
@Premduth Could you please give me some help?

ombhilare999, 2022-03-05:
@allyzhou when will be vitis flow supporting vck5000 es1 card?

Premduth, 2022-03-07:
There are no plans to support a Vitis flow for the es1 card. In order to use the Vitis flow you have to be on the VCK5000 production card.

rahulsingh701, 2022-03-07:
@Premduth when you mean no plans to support Vitis flow for the ES1 card, do you mean there will be no examples provided for it? If I remember correctly the ES1 card supports Vitis 2020.2 and there are plans for updated shells which will allow it to be used with Vitis 2021.1.1 .

Premduth, 2022-03-07:
Correct no new Vitis examples will be provided for the es1 card.

rahulsingh701, 2022-03-07:
@Premduth  So just confirming it is possible to access the AIE kernel and implement Hardware accelerated Pre-Post processing with Vitis 2020.2 but there are no tutorials present for it. Is this correct?

allyzhou, 2022-03-09:
no plan to update ES1 platform to make it work with Vitis newer version, and we have no plan to provide any Vitis end-to-end design examples on VCK5000 ES1 platform.

ombhilare999, 2022-03-09:
okay does that mean using the current shell I can try accessing AIE cores using vitis 2020.2 version?
Can we follow vck5000 or alveo card end to end example and replicate the results for vck5000 es1 card on current shell + vitis 2020.2 version?

allyzhou, 2022-03-09:
I don't think your shell version can work with 2020.2 Vitis, if you're using Vitis AI, the download page will provide you the detailed Vitis, XRT, and platform version, please strictly follow that. Other than what we announced, we have no plan and no resource for enough regression test run to guarantee the other combinations of tools and platforms will work.

gustandil, 2022-04-20:
Hi, I'm trying to understand the limitation of ES1 boards.
We installed a few months ago and test VCK5000-ES1. 
Now we tried to reproduce the examples, and the reference from lounge was "disappeared".
I googled without results of the former information. Is available?

Best Regards, GS

Premduth, 2022-04-20:
VCK5000-ES1 cards and pretty much any ES1 cards are only supported via lounge access. Please contact your nearest sales office for instructions to the lounge.