Xilinx/Vitis-Tutorials #156: Getting Started Vitis Example U50 hw error

nediver, 2022-01-12:
Hello. I'm working on an example from the link below.

https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Getting_Started/Vitis/Part4-data_center.md

However, when I run the following command, the following error occurs.

~/Vitis-Tutorials-2021.2/Getting_Started/Vitis/example/u50/hw$ ./app.exe
terminate called after throwing an instance of 'xrt_xocl::error'
  what():  No devices found
Aborted (core dumped)

As shown below, the installation of U50 seems to have gone well.

Card [0000:02:00.0]
    Card type:          u50
    Flash type:         SPI
    Flashable partition running on FPGA:
        xilinx_u50_gen3x16_xdma_201920_3,[ID=0xf465b0a3ae8c64f6],[SC=5.2.15]
    Flashable partitions installed in system:
        xilinx_u50_gen3x16_xdma_201920_3,[ID=0xf465b0a3ae8c64f6],[SC=5.2.15]

Both sw_emu and hw_emu are cleared, and a problem occurs in executing the last app.exe in the hw stage.

I've done everything from installation to validation of U50. Does anyone know a solution?

allyzhou, 2022-01-13:
dmesg log should be able to give you some hit on what's wrong. BTW, did you examine the U50 using xbutil validate?

imrickysu, 2022-02-24:
Since there are no active following up posts, I'm closing this issue. If you still need help in this issue, please reopen this issue.

closed 2022-02-24 by imrickysu