Xilinx/Vitis-Tutorials #163: launch_emulator: command not found

Candice-Yin, 2022-01-14:
When I input the order "./launch_hw_emu.sh", it dispay as follows:
[image]

randyh62, 2022-01-14:
It looks like the `$PATH` is not setup properly as the `launch_emulator` command should be accessible from the Vitis installation. This is usually configured when the environment is setup with the `settings.sh` script. The `launch_hw_emu.sh `script is generated by the v++ command during the build process, and should not need to be run under `sudo`.

Candice-Yin, 2022-01-15:
@randyh62 Hi! I set up the $PATH as follows:
[image]
[image]
The strange thing is that I succeeded in this step before.

randyh62, 2022-01-16:
It looks like your setup is correct. Not sure why it would work one time and then not the next.  Can you type "which launch emulator" and see what is returned?

Candice-Yin, 2022-01-17:
> It looks like your setup is correct. Not sure why it would work one time and not the next. Can you type "which launch_emulator" and see what is returned?

I find that the problem is about user rights. Thank you. @randyh62

closed 2022-01-17 by Candice-Yin