Xilinx/Vitis-Tutorials #164: HW Boot Issue - 01-Versal_Custom_Thin_Platform_Extensible_System - 2021.1

paulscooper, 2022-01-14:
https://github.com/Xilinx/Vitis-Tutorials/tree/2021.1/Developer_Contributed

This project is almost completely script based so I am struggling to see where I went wrong. 
I dd the sd_card.img to my SD card, and when I boot it hangs after about 10s.

Anyone else experience this ?
If its not just me, anyone have a fix?

Thanks

INIT: version 2.97 booting                                                      
Starting udev                                                                   
[    6.799041] udevd[222]: starting version 3.2.9                               
[    6.803946] random: udevd: uninitialized urandom read (16 bytes read)        
[    6.810420] random: udevd: uninitialized urandom read (16 bytes read)        
[    6.816883] random: udevd: uninitialized urandom read (16 bytes read)        
[    6.827137] udevd[223]: starting eudev-3.2.9                                 
[    6.873240] zocl: loading out-of-tree module taints kernel.                  
[    6.891921] [drm] Probing for xlnx,zocl-versal                               
[    6.896695] zocl-drm amba_pl@0:zyxclmm_drm: IRQ index 32 not found           
[    6.902944] [drm] FPGA programming device versal_fpga founded.               
[    6.908779] [drm] PR Isolation addr 0x0                                      
[    6.986273] [drm] Initialized zocl 0.0.0 00000 for amba_pl@0:zyxclmm_drm on 0
[    7.127662] random: fast init done                                           
[    7.153448] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data.
[    7.599652] zocl-drm amba_pl@0:zyxclmm_drm: IRQ index 32 not found

BrunoVandeveldeXilinx, 2022-01-15:
Be aware that default it builds for an es1-board, so you need to modify a line in the main Makefile (see README.md) for pre-production board!

paulscooper, 2022-01-17:
Yep, I had already changed this and build for pre prod board. Can I ask it you have tried it with 2021.1? 

I won't get back to it until Tuesday, but will re-do the entire process from scratch. Apart from Linux build its pretty fast :)

BrunoVandeveldeXilinx, 2022-01-18:
Yes, the tutorial has been created in 2021.1, and has been tested a lot with that version.
If you have the option to use 2021.2, then it's even better, since some clocking issues where resolved within that version!
You can find the 2021.2 tutorial in: https://github.com/Xilinx/Vitis-Tutorials/tree/2021.2/Developer_Contributed/01-Versal_Custom_Thin_Platform_Extensible_System

Indeed, the Linux build takes a while, since we build the whole Linux environment, and also the sysroot.
Nevertheless the idea is that once the HW/SW-platform is stable, you don't need to re-build it anymore :-) 
Then you only need to re-build the Vitis part after modifications to AIE and/or RTL/HLS kernels.

Premduth, 2022-01-19:
@paulscooper believe you are already doing this, although wanted to confirm. When targeting 2021.1, make sure you are changing the branch to 2012.1.. The default is 2021.2 and the 2021.2 branch is not compatible with the 2021.1 release.. We regress the branches against the releases.. 

In addition agreed with @BrunoVandeveldeXilinx . If possible it is better to switch to 2021.2 as there are a lot of good improvements for AIE designs in 2021.2

BrunoVandeveldeXilinx, 2022-01-19:
@Premduth In the tutorial we check the version (Petalinux, Vivado and Vitis)! As long as @paulscooper did not disable that, should be no problem.

BrunoVandeveldeXilinx, 2022-03-16:
@paulscooper can you close this item if you are not longer affected or no longer need this? Thanks!

closed 2022-03-22 by imrickysu