Xilinx/Vitis-Tutorials #186: g++: error: unrecognized command line option ‘-std=c++14’

jiajiahao0619, 2022-02-14:
I am running the 2021.2 branch. 
It is normal until run the make host command. It is so kindly that tell me how to fix it?

[image]

imrickysu, 2022-02-14:
Hi @jiajiahao0619 , Vitis doesn't ship `g++` in its installer. Host application compilation uses the system default GCC/G++.
It's possible that on CentOS/RHEL, default GCC doesn't support C++14. If you'd like to use `-std=c++14`, please upgrade your system GCC version.

imrickysu, 2022-02-21:
Hi @jiajiahao0619 , BTW, if you're working on an embedded application, this error may indicate that you didn't setup SYSROOT properly. Please check the environment setup guidance of the tutorial you are referring to.

jiajiahao0619, 2022-02-21:
Thanks @jiajiahao0619  It is really helpful for me. it is normal now.

closed 2022-02-21 by imrickysu