Xilinx/Vitis-Tutorials #193: Error Vitis Analyzer - tutorial: 13-aie-performance-analysis

nielsvanetten, 2022-02-24:
https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/AI_Engine_Development/Feature_Tutorials/13-aie-performance-analysis/aie_execution_measurement.md

When opening the trace file tap in vitis_analyzer I receive the error: /tmp/vitis_analyzer/[user]/system.wdb is not a valid trace report file.
Using fedora release 34.

nielsvanetten, 2022-02-24:
seems there is some weird mapping going on, if i make that directory and put the system.wdb and .wcfg files in that directory it works. Is there any way to configure the vitis_analyzer where it looks for these files?

AlphaBu, 2022-02-25:
Hi, @nielsvanetten Could you try run following command before opening Vitis Analyzer: 
**vcdanalyze --vcd=foo.vcd --wdb --ctf**
Then see if this issue is still there? 
It can not be reproduced for me. Not sure if it's related to Linux version.

nielsvanetten, 2022-02-28:
it work when i do the command **vcdanalyze --vcd=foo.vcd --wdb** and then put the files in this tmp folder. Not sure why vitis analyzer is looking in that folder for the file and how i can change it.

nielsvanetten, 2022-03-02:
that did the trick. Thank you!

closed 2022-03-02 by nielsvanetten