Xilinx/Vitis-Tutorials #210: LeNet Versal AIE simulation stalls

JieGH, 2022-03-09:
Hi,

I build and ran the AI Engine simulation for the cores used in https://github.com/Xilinx/Vitis-Tutorials/tree/2021.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial

The build was successful, run output matched with reference. However, the core execution never end, program stall at main. 

Did anyone also experience this ?

```
``
aiesimulator --pkg-dir=./Work --i=.. --dump-vcd=foo 
AIEsim feature license is found.
INFO: Executing config: ./Work/config/scsim_config.json
Info: If you are running trace or profile, Xilinx recommends to use --xlopt=0. This will give a greater visibility into the design. One can verify --xlopt level in the aiecompiler command line.
Loading device config from: /tools/Xilinx/Vitis/2021.2/aietools/data/devices/VC1902.json
Initializing AIE driver...
Initializing ADF API...
IP-INFO: [ps_i33_ps_main] IP loaded.

Warning: AieSimConfig: xv_cpt_lib_path env is not set
In file: aie_config.cpp:451

Warning: AieSimConfig: xv_cxl_ip_path env is not set
In file: aie_config.cpp:472

Info: DEVICE FILE: /tools/Xilinx/Vitis/2021.2/aietools/data/devices/VC1902.json

Info: AIE SOLUTION FILE: ./Work/arch/aieshim_solution.aiesol
[AIESIM_OPTIONS]: aiesim_options file path aiesimulator_output/aiesim_options.txt
Disable unused tiles
xpe directory: ./Work
Found xpe file ./Work/reports/graph.xpe
ISS disables unused tiles
Optimized ISS Wrapper (R+W), r2p19
Running Dispatch Server on port: 34189
Current Directory=/home/user/user_doc/userFolder/ws/aie_mac_2_2core1matrix/Emulation-AIE
--------------------------------------------------------------------------------------------------
SLAVE STREAM INFO:  | Port Name         | Port Width        | Stream Type       | Channel Index     
                    | S00_AXIS          | 128               | PL STREAM         | 56, 57            
                    | S01_AXIS          | 128               | PL STREAM         | 48, 49            
                    | S02_AXIS          | 128               | PL STREAM         | 52, 53            
                    | S03_AXIS          | 128               | PL STREAM         | 104, 105          
                    | S04_AXIS          | 128               | PL STREAM         | 120, 121          
                    | S05_AXIS          | 128               | PL STREAM         | 136, 137          
                    | S06_AXIS          | 128               | PL STREAM         | 152, 153          
                    | S07_AXIS          | 128               | PL STREAM         | 176, 177          
                    | S08_AXIS          | 128               | PL STREAM         | 192, 193          
                    | S09_AXIS          | 128               | PL STREAM         | 224, 225          
                    | S10_AXIS          | 128               | PL STREAM         | 256, 257          
--------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------
MASTER STREAM INFO: | Port Name         | Port Width        | Stream Type       | Channel Index     
                    | M00_AXIS          | 128               | PL STREAM         | 42, 43            
                    | M01_AXIS          | 128               | PL STREAM         | 38, 39            
                    | M02_AXIS          | 128               | PL STREAM         | 40, 41            
                    | M03_AXIS          | 128               | PL STREAM         | 78, 79            
                    | M04_AXIS          | 128               | PL STREAM         | 90, 91            
                    | M05_AXIS          | 128               | PL STREAM         | 102, 103          
                    | M06_AXIS          | 128               | PL STREAM         | 114, 115          
                    | M07_AXIS          | 128               | PL STREAM         | 138, 139          
                    | M08_AXIS          | 128               | PL STREAM         | 180, 181          
--------------------------------------------------------------------------------------------------

Warning: (W505) object already exists: tl.logicalNoC.PS_AIE_Config_Master.delayed_event. Latter declaration will be renamed to tl.logicalNoC.PS_AIE_Config_Master.delayed_event_0
In file: sc_object_manager.cpp:149
TraceManager type=psql_db
Enabled fast PM writes.
Enabled fast DM writes.
 [warning] : TCL interface disabled.

Warning: (W505) object already exists: tl.aie_logical.aie_xtlm.math_engine.clockM. Latter declaration will be renamed to tl.aie_logical.aie_xtlm.math_engine.clockM_0
In file: sc_object_manager.cpp:149

Warning: (W505) object already exists: tl.aie_logical.aie_xtlm.math_engine.resetM. Latter declaration will be renamed to tl.aie_logical.aie_xtlm.math_engine.resetM_0
In file: sc_object_manager.cpp:149
NOCSlaveUnitXtlm::Detail::setAxiPhySize(): NOC_NSU128_X8Y6, reg_axi_width_byte=64B > physical interface=16B
NOCSlaveUnitXtlm::Detail::setAxiPhySize(): NOC_NSU512_X0Y0, reg_axi_width_byte=64B > physical interface=64B
TraceManagerDB::startTrace
Enable stream switch latency
Starting to send data from file: ../data/0_1/matA_in_128plio.txt
Starting to send data from file: ../data/0_1_1/matA_in_128plio.txt
Starting to send data from file: ../data/0_1_2/matA_in_128plio.txt
Starting to send data from file: ../data/0_1_3/matA_in_128plio.txt
Starting to send data from file: ../data/0_1_4/matA_in_128plio.txt
Starting to send data from file: ../data/0_1_5/matA_in_128plio.txt
Starting to send data from file: ../data/0_1_6/matA_in_128plio.txt
Starting to send data from file: ../data/0_3/matA_in_128plio.txt
Starting to send data from file: ../data/0_5/matA_in_128plio.txt
Starting to send data from file: ../data/0_3_0/matA_in_128plio.txt
Starting to send data from file: ../data/0_5_0/matA_in_128plio.txt
Starting to receive data into file: aiesimulator_output/output01.txt
Starting to receive data into file: aiesimulator_output/output01_1.txt
Starting to receive data into file: aiesimulator_output/output01_2.txt
Starting to receive data into file: aiesimulator_output/output01_3.txt
Starting to receive data into file: aiesimulator_output/output01_4.txt
Starting to receive data into file: aiesimulator_output/output01_5.txt
Starting to receive data into file: aiesimulator_output/output01_6.txt
Starting to receive data into file: aiesimulator_output/output04.txt
Starting to receive data into file: aiesimulator_output/output04_0.txt
IP-INFO: [ps_i33_ps_main] IP started.

Info: (I702) default timescale unit used for tracing: 1 ps (foo.vcd)
Loading elfs of graph g...
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
 [warning] : TCL interface disabled.
Initializing graph g...
Resetting cores of graph g...
Configuring DMAs of graph g...
Configuring PL-Interface for graph g...
Calling adf::graph::run() without specifying number of iterations ... It will either run AIE cores indefinitely, or run AIE cores for the last specified iterations if adf::graph::run(iterations) was called before.
Enabling core(s) of graph g
Waiting for core(s) of graph g to finish execution ...
``
```

xflorentw, 2022-03-14:
What OS are you using?
How long did you wait for the simulation to complete?

Also, you are only running AIE simulation right? Is it using the makefile? I do not see the option... (so this is not supported officially if not in makefile)

If this is not running the makefile, then you can see in graph.cpp that there is not iterations set in graph.run() so the graph will run indefinitely. You need to add a timeout if you want the simulation to stop:
--simulation-cycle-timeout=50000

xflorentw, 2022-03-15:
Hi @JieGH 

You might have forgotten to rebuild the aie application after adding 1 interation for graph.run() in graph.cpp because this is working for me with it. The simulation is exiting

closed 2022-03-15 by JieGH