Xilinx/Vitis-Tutorials #214: Hardware_Acceleration

xooxit, 2022-03-16:
While building the `Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes` project, I got some errors at the stage of FPGA logic optimization. My vitis version is 2021.2 and Ubuntu version is 18.04. 

I found some advice related to the Error below, and they said created HDL wrapper, but hmm.. anyway, is there any guy who gets a similar error like this?

```
****** vpl v2021.2 (64-bit)
  **** SW Build 3363252 on 2021-10-14-04:41:01
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/home/lab/yong/document/Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes/_x/link/int/kernel_info.dat'.
INFO: [VPL 74-78] Compiler Version string: 2021.2
INFO: [VPL 60-423]   Target device: xilinx_u250_xdma_201830_2
INFO: [VPL 60-1032] Extracting hardware platform to /home/lab/yong/document/Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes/_x/link/vivado/vpl/.local/hw_platform
[15:25:20] Run vpl: Step create_project: Started
Creating Vivado project.
[15:25:30] Run vpl: Step create_project: Completed
[15:25:30] Run vpl: Step create_bd: Started
[15:26:46] Run vpl: Step create_bd: RUNNING...
[15:27:02] Run vpl: Step create_bd: Completed
[15:27:02] Run vpl: Step update_bd: Started
[15:27:03] Run vpl: Step update_bd: Completed
[15:27:03] Run vpl: Step generate_target: Started
[15:28:19] Run vpl: Step generate_target: RUNNING...
[15:29:35] Run vpl: Step generate_target: RUNNING...
[15:30:17] Run vpl: Step generate_target: Completed
[15:30:17] Run vpl: Step config_hw_runs: Started
[15:30:21] Run vpl: Step config_hw_runs: Completed
[15:30:21] Run vpl: Step synth: Started
[15:30:52] Block-level synthesis in progress, 0 of 8 jobs complete, 1 job running.
[15:31:22] Block-level synthesis in progress, 0 of 8 jobs complete, 1 job running.
[15:31:53] Block-level synthesis in progress, 0 of 8 jobs complete, 1 job running.
[15:32:23] Block-level synthesis in progress, 0 of 8 jobs complete, 1 job running.
[15:32:44] Run vpl: Step synth: Completed
[15:32:44] Run vpl: Step impl: Started
[15:45:21] Finished 2nd of 6 tasks (FPGA linking synthesized kernels to platform). Elapsed time: 00h 20m 42s 

[15:45:21] Starting logic optimization..
[15:46:51] Phase 1 Generate And Synthesize MIG Cores
[15:51:24] Phase 2 Generate And Synthesize Debug Cores
[15:53:25] Phase 3 Retarget
[15:53:55] Phase 4 Constant propagation
[15:53:55] Phase 5 Sweep
[15:54:56] Phase 6 BUFG optimization
[15:55:26] Phase 7 Shift Register Optimization
[15:55:26] Phase 8 Post Processing Netlist
[15:56:39] Run vpl: Step impl: Failed
[15:56:39] Run vpl: FINISHED. Run Status: impl ERROR

===>The following messages were generated while processing /home/lab/yong/document/Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes/_x/link/vivado/vpl/prj/prj.runs/impl_1 :
ERROR: [VPL 101-2] ERROR: [Common 17-55] 'set_property' expects at least one object.
Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.
ERROR: [VPL 101-3] sourcing script /home/lab/yong/document/Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes/_x/link/vivado/vpl/scripts/impl_1/_full_place_pre.tcl failed
ERROR: [VPL 60-773] In '/home/lab/yong/document/Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes/_x/link/vivado/vpl/runme.log', caught Tcl error:  problem implementing dynamic region, impl_1: place_design ERROR, please look at the run log file '/home/lab/yong/document/Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes/_x/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: place_design ERROR, please look at the run log file '/home/lab/yong/document/Vitis-Tutorials/Hardware_Acceleration/Design_Tutorials/05-bottom_up_rtl_kernel/krnl_aes/_x/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [15:56:40] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:40 ; elapsed = 00:32:04 . Memory (MB): peak = 2071.566 ; gain = 0.000 ; free physical = 83290 ; free virtual = 99154
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:81: recipe for target 'build_hw' failed
make: *** [build_hw] Error 1
```

imrickysu, 2022-03-21:
Hi @Rampagee , could you please take a look and check whether you can reproduce this issue?
It seems some properties is empty in the tcl script `impl_1/_full_place_pre.tcl`.

Rampagee, 2022-03-22:
Hi, @xooxit , sorry, I cannot re-produce the errors with same configuration. Below is my relevant log segment. Would you please help to confirm whether the _Makefile_ and _krnl_aes_test.xdc_ files are modified according to U250 case?

[15:50:26] Block-level synthesis in progress, 81 of 82 jobs complete, 0 jobs running.
[15:50:56] Block-level synthesis in progress, 81 of 82 jobs complete, 1 job running.
[15:51:26] Block-level synthesis in progress, 82 of 82 jobs complete, 0 jobs running.
[15:51:57] Top-level synthesis in progress.
[15:52:27] Top-level synthesis in progress.
[15:52:54] Run vpl: Step synth: Completed
[15:52:54] Run vpl: Step impl: Started
[16:00:29] Finished 2nd of 6 tasks (FPGA linking synthesized kernels to platform). Elapsed time: 00h 23m 30s 

[16:00:29] Starting logic optimization..
[16:00:59] Phase 1 Generate And Synthesize MIG Cores
[16:05:03] Phase 2 Generate And Synthesize Debug Cores
[16:06:34] Phase 3 Retarget
[16:06:34] Phase 4 Constant propagation
[16:06:34] Phase 5 Sweep
[16:07:04] Phase 6 BUFG optimization
[16:07:34] Phase 7 Shift Register Optimization
[16:07:34] Phase 8 Post Processing Netlist
[16:09:36] Finished 3rd of 6 tasks (FPGA logic optimization). Elapsed time: 00h 09m 06s 

[16:09:36] Starting logic placement..
[16:10:06] Phase 1 Placer Initialization
[16:10:06] Phase 1.1 Placer Initialization Netlist Sorting
[16:13:39] Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device

AnusheelXilinx, 2024-12-23:
With the recent updates in the tutorial, we do not see this issue internally. 
Please feel free to test the 2023.2 version and let us know if you still see an issue.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx