Xilinx/Vitis-Tutorials #215: explanation confusing example: 02-using-gmio

nielsvanetten, 2022-03-17:
At the bottem of: https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/AI_Engine_Development/Feature_Tutorials/02-using-gmio/single_aie_gmio.md

Its stated:
```
After the package is done, run the following commands in the Linux prompt after booting Linux from an SD card:

export XILINX_XRT=/usr
cd /mnt/sd-mmcblk0p1
./host.exe a.xclbin
```
When working with the VCK190, do they mean booting the VCK190 with a petalinux version installed on the SD card and then running these commands from your own Linux PC ?

imrickysu, 2022-03-21:
Please run these commands on VCK190 Linux.

imrickysu, 2022-03-21:
Hi @AlphaBu , `export XILINX_XRT=/usr` is not needed for embedded acceleration. Could you update the instructions while clarifying this issue?

closed 2022-03-21 by nielsvanetten