Xilinx/Vitis-Tutorials #219: Problem while Hardware_Accelerators\Introduction\03-Algorithm_Acceleration\docs\module1_baseline

AzzHub, 2022-03-20:
I import the project according to the readme. I can run Emulation-SW correctly and soonly,  while the Emulation-HW is very slow even longer than ten hours!
The only step defferent to the readme is I add  Program Arguments : `-xclbin /home/module1baseline/cholesky_system/Emulation-SW/binary_container_1.xclbin`. 
PS1: Emulation-HW is very slow even I add  Argument:  `-M 1 -N 1`.
PS2: The Debug can't run correctly while Emulation-HW is active with showing 

> No source available for "main() at 0x555555556c10"

AzzHub, 2022-03-20:
I added some printf for checking, and found that:
    the program stalls at `q.finish()` in line 172(in "test.cpp")
`    // Data transfer from host buffer to device buffer
    std::vector<std::vector<cl::Event> > kernel_evt(2);
    kernel_evt[0].resize(1);
    kernel_evt[1].resize(1);

    std::vector<cl::Memory> ob_io;
    ob_io.push_back(buffer[0]);

    q.enqueueMigrateMemObjects(ob_io, 0, nullptr, &kernel_evt[0][0]); // 0 : migrate from host to dev
    q.finish();
    std::cout << "INFO: Finish data transfer from host to device" << std::endl;`

randyh62, 2022-03-21:
I have tested it and it took about 15 minutes to complete the hw_emu run as you can see from the attached screenshot. Notice that emconfigutil ran at 2:40 and the emulation completed by 3:00 pm. The build itself took one hour 8 minutes as you can also see reported prior to emconfigutil. 

Perhaps there is an issue with your system memory or OS? Can you provide a few details about the configuration of your environment?
[image]

randyh62, 2022-03-22:
Since you are new to Vitis, perhaps you could start with the https://github.com/Xilinx/Vitis-Tutorials/tree/2021.2/Getting_Started/Vitis tutorial? Focusing on the U200 for the Data Center acceleration card. See if you can get the hw_emu build to run for that tutorial?

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx