Xilinx/Vitis-Tutorials #220: 2021.2 - AIE - design tutorial - 09 post-link-recompile - VitisIDE_flow . step 16 needs updating?

delorytheape, 2022-03-22:
Hi,
While following https://github.com/Xilinx/Vitis-Tutorials/tree/2021.2/AI_Engine_Development/Design_Tutorials/09-post-link-recompile/VitisIDE_Flow.md, it seems that Phase1, step 16 may need updating (??) 
ie. For Vitis 2021.2 there appears to be no column "PL Kernel in AIE Graph" in the "Hardware Functions" of the PL project (unless I have done something incorrect ?? but I think I have followed everything else correctly).
When I try to build the whole project (Cosim1_system_hw_link [pl] | Build Project), it fails with various errors including;
ERROR: [CFGEN 83-2284] No stream resources found that can accomodate compute unit "polar_clip_1.in_sample"
which I assume is related to step 16  that can no longer be performed. 

The makefile version of the tutorial completed without error.

delorytheape, 2022-03-22:
For anyone interested: The template example included in 2021.2 (AI Engine, PL and PS System Design) is basically the same, but works correctly. The template example uses a slightly different graph that does not include the PL kernel 'polar_clip', but rather has two additional ports to connect to the PL kernel.

OTremois, 2022-03-31:
The Lab has been completely revisited in 2022.1.
polar_clip is out of the graph
A new use case has been added which does not need an intermediate platform to be created.

closed 2022-04-20 by imrickysu