Xilinx/Vitis-Tutorials #226: versal experiment: 01-aie_a_to_z - multiple graphs

nielsvanetten, 2022-03-31:
For some experiment with the VCK190 I'm trying to maximize the amount of parallel graph I can connect to the AI engine.
However, when modifying the main.cpp from the example aie_a_to_z, it gets stuck when running on the vck190 after 4 OR 5 graphs (every run different).
I observed that the problem is that there seems to be no data coming in, in the s2mm block, so it gets stuck in the while loop at:
`    while(1) {
    	printf("Checking output, check #%d\n",checks);
    	uint32_t v = Xil_In32(S2MM_BASE_4 + CTRL_OFFSET); // or at S2MM_BASE_5
    	checks++;
    	if(v & 6) {
    		break;
    	}
    	sleep(2);
    }`

Note that I did check the addresses of the mm2s / s2mm in Vivado, these should be correct.
See my code in the attachement. 
[main.txt](https://github.com/Xilinx/Vitis-Tutorials/files/8390657/main.txt)

Note that if I for example uncomment graph 0 to 5 for example, graph 6 to 10 OR 11 do run and than it gets stuck at graph 11 or 12.

Thanks in advance for your help!

xflorentw, 2022-04-01:
Hi @nielsvanetten 

This is not much related to the A to Z example. Please create a question on the forums with the topic [AI ENGINE ARCHITECTURE & TOOLS](https://support.xilinx.com/s/topic/0TO2E000000YKXjWAO/ai-engine-architecture-tools) for your question.

Github issues are for issues with the design itself or tutorial content.

nielsvanetten, 2022-04-01:
thanks, will do

closed 2022-04-01 by nielsvanetten