Xilinx/Vitis-Tutorials #227: vitis 2021.2 custom platform cannot boot with DP monitor

bok960731, 2022-04-01:
By referring to the zcu104 custom platform creation of the Vitis Design Tutorial, 
the custom board was operated normally using the 2020.2 Tool and Vitis AI 1.3.

To use Vitis AI 1.4, the Tool version was updated to 2021.2, and an image with a new platform creation and 
DPU added was also created by referring to the master branch of custom platform creation.

The problem is, booting through a serial connection like putty works well, 
but the DP monitor connected to the board does not work like when using the previous 2020.2.

The picture below is the screen of the DP Monitor connected to the board I want.

[image]

Whether it is vadd example or DPU example, the screen does not appear on the monitor connected to the board. Is there a solution?

sven2314, 2022-04-11:
@bok960731 
Hi, could you help to confirm whether the application runs and log output is as expected. Besides to help to analyze the problem It would be better to share the boot log  and application output log from UART  terminal and also the hardware and software ENV to us.

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx