Xilinx/Vitis-Tutorials #233: Possible Data Corruption in Example Design_Tutorials/06-fft2d_AIEvsHLS

K-Wu, 2022-04-13:
When running `aiesimulator` on `Design_Tutorials/06-fft2d_AIEvsHLS`, I continue to get the following messages.

> WARNING : Line skipped, invalid number of data samples on line, got 1 expected 8

To work this around, I replaced `1.0` with `1` in the two input files `Design_Tutorials/06-fft2d_AIEvsHLS/AIE/design/aie_src/aiesim_data/input0.txt` and `Design_Tutorials/06-fft2d_AIEvsHLS/AIE/design/aie_src/aiesim_data/input1.txt`. This suppressed the aforementioned warning messages.

I still want to check if the dataset files content is correct and understand the best way to fix this. Thanks!

imrickysu, 2022-04-20:
Hi @JithinPillai, could you help with this question?

JithinPillai, 2022-04-20:
Hi @imrickysu,
It is happening due to data being provided for the cint16 design is same for cfloat, I will fix that

JithinPillai, 2022-06-21:
Hi @K-Wu,
The issue should be fixed now!
Sorry for the delay the I was out of office due to some personal matters.
Please have a look.

Regards
Jithin

K-Wu, 2022-06-21:
Thank you @JithinPillai  for addressing this issue! I have stopped working on the project using this for a while but I also believe we can close this issue now as it is fixed. Thanks again!

closed 2022-06-21 by K-Wu