Xilinx/Vitis-Tutorials #234: Question about buffer syncing

ledwmp, 2022-04-15:
Hi, I'm running into a problem and not sure if it's on my end.

Trying to run hardware emulation in vivado 2021.1 similar to the vadd example with a user-managed kernel and the cpp API. 

When I try to sync a buffer from host to DDR it looks like I'm syncing an empty buffer, even though I fill the host pointer. 

Similar to this:

`
//initialize device
auto bf = xrt::xclbin("./my.xclbin");
auto device = xrt::device(0);
auto uuid = device.load_xclbin(bf);
auto myip = xrt::ip(device,uuid,"mykernel");

//create 4k-aligned buffer
unsigned char * hostptr;
int ret = posix_memalign(&hostptr, 4096, sizeofbuffer); 

//fill buffer and check that it is indeed filled

//create buffer object with host pointer
auto buff = xrt::bo(device, hostptr, sizeofbuffer, 1);
buff.sync(XCL_BO_SYNC_BO_TO_DEVICE);
uint64_t addr = bo.address();

//write registers and start
myip.write_register(register,addr);
myip.write_register(register + 4, addr >> 32); 
myip.write_register(user_ctrl, start);
`

It seems fairly straightforward, but when I look at the data streaming out of that address it's just empty. If I point to another random address in memory I can stream random garbage, and I know that I'm streaming out of the correct address because the starting stream address is the same as the one I wrote to the register. Where am I going wrong here?

Thanks!

I

randyh62, 2022-04-15:
It is hard to tell what is going on in your pseudocode. Something could be a typo or could be an error. 

I applied your code to the RTL kernel tutorial you seem to be referring to, and was able to build and run the code after sorting through a few errors:
https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Hardware_Acceleration/Feature_Tutorials/01-rtl_kernel_workflow/reference-files/src/host/user-host.cpp

this line seems wrong:
`uint64_t addr = bo.address();`
It should point to the 'buff' object from this line:
`auto buff = xrt::bo(device, hostptr, sizeofbuffer, 1);`

And in my test posix_memalign was expecting a void pointer for &hostptr, but presumably you got your code to build without error? In any case, your host code would be more revealing than the pseudocode.

ledwmp, 2022-04-15:
Hey, thanks for getting back. My bad, I threw together the pseudocode quickly and you're right, I meant to call address on the created buffer object. I've tried the map method, creating the buffer without a pointer and then using the write method to fill the buffer, and all return the same result.

I'll put together a reproducible example and see if I can recreate the problem in a minimal setting and get back to you.

Thanks!

ledwmp, 2022-04-21:
I was not able to recreate this problem with a simple system consisting of a single master and slave. I think there is a problem somewhere in my IP where data are not being transferred despite following AXI protocol specifications. I'll have to dig in deeper, sorry for the inconvenience!

closed 2022-04-21 by ledwmp