Xilinx/Vitis-Tutorials #255: libboost-dev package dependency on Ubuntu 18.04 to compile Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/01-rtl_kernel_workflow with xilinx_u280_xdma_201920_3

fergusonml, 2022-06-06:
$ xbmgmt examine
System Configuration
  OS Name              : Linux
  Release              : 4.15.0-180-generic
  Version              : #189-Ubuntu SMP Wed May 18 14:13:57 UTC 2022
  Machine              : x86_64
  CPU Cores            : 4
  Memory               : 32129 MB
  Distribution         : Ubuntu 18.04.6 LTS
  GLIBC                : 2.27
  Model                : MS-7971

XRT
  Version              : 2.11.634
  Branch               : 2021.1
  Hash                 : 5ad5998d67080f00bca5bf15b3838cf35e0a7b26
  Hash Date            : 2021-06-08 22:10:49
  XOCL                 : 2.11.634, 5ad5998d67080f00bca5bf15b3838cf35e0a7b26
  XCLMGMT              : 2.11.634, 5ad5998d67080f00bca5bf15b3838cf35e0a7b26

Devices present
  [0000:01:00.0] : xilinx_u280_xdma_201920_3 
*****************************************************************************
compile error:
g++ -g -I./ -I/opt/xilinx/xrt/include -I~/Xilinx/Vivado/2021.1/include -Wall -O0 -g -std=c++1y -fmessage-length=0 ./src/host/user-host.cpp -o 'host' -L/opt/xilinx/xrt/lib -lxrt_coreutil -pthread -lrt -lstdc++
In file included from ./src/host/user-host.cpp:22:0:
/opt/xilinx/xrt/include/xrt/xrt_device.h:28:11: fatal error: boost/any.hpp: No such file or directory
 # include <boost/any.hpp> // std::any c++17
           ^~~~~~~~~~~~~~~
compilation terminated.
Makefile:118: recipe for target 'host' failed
make: *** [host] Error 1
*****************************************************************************
sudo apt install libboost-dev
resolved issue.
see: https://github.com/Xilinx/XRT/issues/6295

imrickysu, 2022-06-14:
Hi @randyh62 , please update compile script to use c++17. This is an updated requirement in XRT 2022.1.
Ref: https://github.com/Xilinx/XRT/blob/master/CHANGELOG.rst#2130-202210213x

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx