Xilinx/Vitis-Tutorials #266: The Program xchesscc has encountered an unexpected error exiting ... xchesscc Failed No such device

HadXu, 2022-06-28:
The Program xchesscc has encountered an unexpected error exiting ... xchesscc Failed No such device.
But lspci show device below. 
```bash
18:00.0 Memory controller: Xilinx Corporation Device 5048
        Subsystem: Xilinx Corporation Device 000e
        Physical Slot: 1
        Flags: bus master, fast devsel, latency 0, NUMA node 0
        Memory at 387ff0000000 (64-bit, prefetchable) [size=128M]
        Memory at 387ff8000000 (64-bit, prefetchable) [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: xclmgmt
        Kernel modules: xclmgmt

18:00.1 Memory controller: Xilinx Corporation Device 5049
        Subsystem: Xilinx Corporation Device 000e
        Physical Slot: 1
        Flags: bus master, fast devsel, latency 0, NUMA node 0
        Memory at 387ff8020000 (64-bit, prefetchable) [size=64K]
        Memory at 387fe0000000 (64-bit, prefetchable) [size=256M]
        Capabilities: <access denied>
        Kernel driver in use: xocl
        Kernel modules: xocl
```

xflorentw, 2022-06-28:
What tutorial are you trying to use?
What command did you run?

HadXu, 2022-06-29:
I use [Vitis_2021.2](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vitis.html) software. [mlir-aie](https://github.com/Xilinx/mlir-aie) need aietools/bin/xchesscc.

xflorentw, 2022-06-30:
If you are using [mlir-aie](https://github.com/Xilinx/mlir-aie) why not creating the issue under [mlir-aie](https://github.com/Xilinx/mlir-aie) repo?
This way the right people can look at it

HadXu, 2022-06-30:
Ok. Sorry.