Xilinx/Vitis-Tutorials #270: Unable to generate bitstream for Ethernet core alveo u250

arishsatheesan, 2022-07-08:
Hi,

I am trying to run the tutorial https://github.com/Xilinx/Vitis-Tutorials/tree/2022.1/Hardware_Acceleration/Feature_Tutorials/09-using-ethernet-on-alveo with alveo u250. 

I have used the evaluation licenses given in https://www.xilinx.com/products/intellectual-property/ef-di-25gemac.html, but I am unable to generate the bitstream. The error states that the bitstream generation is not permitted for the core used. Which other license do I need for this tutorial? 

Best regards,

imrickysu, 2022-08-01:
Hi @Rampagee , could you add license expectation to the tutorial? Thanks.

Rampagee, 2022-08-29:
Sorry for late response. Generally, the following two license keys are needed:
xxv_eth_mac_pcs
x_eth_mac

Or refer to following link:
https://www.xilinx.com/products/intellectual-property/ef-di-25gemac/ef-di-25gemac-order.html

If the evaluation licenses are not working, please check the license status in Vivado license manager or consult the licensing solution center... Thanks.

arishsatheesan, 2022-08-29:
> Sorry for late response. Generally, the following two license keys are needed: xxv_eth_mac_pcs x_eth_mac
>
> Or refer to following link: [...]

Thanks for the response.

Both of the above mentioned licences are installed, but still the bitstream generation is failing. 
[image]

Rampagee, 2022-08-30:
Hi, @arishsatheesan, when you apply for the evaluation license from the Xilinx licensing site, you should add both following items, like the snapshot below:
[image]

After importing the generate license file, you should observe the license status in License Manager like below snapshot:
[image]

If this is not the case for you, maybe you need to redo the license generation and importing steps.

arishsatheesan, 2022-09-01:
I generated the license as suggested and imported it. Still did not work.

[image]

Could you please run and check it once to confirm it?

Thanks!

Rampagee, 2022-09-06:
I've tried the license generation flow as mentioned above using a fresh new machine, and the flow can pass. When the generated license is imported with Vivado license manager, whether the following features are avaible in your machine?

[image]

arishsatheesan, 2022-09-06:
Yes, I see all these features in my machine too.

[image]

AnusheelXilinx, 2024-12-23:
If there are any open queries, please feel free to post it in Install and Licensing Community Forums:
https://adaptivesupport.amd.com/s/topic/0TO2E000000YKXwWAO/installation-and-licensing?language=en_US

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx