Xilinx/Vitis-Tutorials #275: AI_engine design tutorials 01-Lenet. Issue in compiling

roseduke3, 2022-07-26:
Hello, 

I'm compiling the 01-Lenet tutorials using the 'make all TARGET=hw_emu' command but an error occurs in the compiling process:

**/home/shininy/Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial/design/host_app_src/lenet_app.cpp:27:10: fatal error: xrt/experimental/xrt_aie.h: No such file or directory**

im now using the platform provided by xilinx vitis and i cant find a file named '**$XILINX_VERSAL/environment-setup-cortexa72-cortexa53-xilinx-linux**'. So i skip that source command in environment setup file. Should this be causing the error? If so, where can I get this file? Any help will be deeply appreciated. 

Best Regards

randyh62, 2022-07-26:
This sounds like an environmental issue. The platform requires some common files to be [installed](https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Installing-Embedded-Platforms?tocId=pm5zCNChp993ACN_ICPFPg) for the XRT runtime. These files can be found in the [Embedded Platforms](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-platforms/2021-2.html) download page.  The **environment-setup-cortexa72-cortexa53-xilinx-linux** should be included under the *Common Images for Embedded Vitis Platforms* section.

roseduke3, 2022-07-26:
> This sounds like an environmental issue. The platform requires some common files to be [...]

imrickysu, 2022-08-01:
Please run through the following steps:
1. Go to https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-platforms/2021-2.html 
2. Download "Versal common image"
3. Extract the downloaded file `tar -xzf xilinx-versal-common-v2021.2.tar.gz` 
4. Extract the sysroot `./sdk.sh`
5. Source environment-setup-cortexa72-cortexa53-xilinx-linux from the extracted sysroot directory.

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns. 

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx