Xilinx/Vitis-Tutorials #278: ERROR: [VPL 41-99] Failed to find BusTerm object axi_ic_processing_system7_0_S_AXI_ACP

chenguoping76, 2022-07-31:
when test Hardware_Acceleration/Feature_Tutorials/01-rtl_kernel_workflow using Makefile:
$make all TARGET=hw DEVICE=<Platform path>/pynqz2.xpfm  (pynqz2.xpfm is a tested custom platform, it can run vadd test case)
when link stage, encount ERROR: [VPL 41-99] Failed to find BusTerm object axi_ic_processing_system7_0_S_AXI_ACP/S01(S00)_AXI_arca(awca,arca)...

randyh62, 2022-07-31:
I am not familiar with the pynqz2.xpfm. Where did you get it from? Can you build the RTL kernel tutorial on another platform, like the zcu102 or zc706?

chenguoping76, 2022-08-01:
We build a custom platform for pynq z2 board according to [Custom Platform Creation on MPSoC](https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104). And pass test the Vadd HLS HW kernel case. 
[image]
[image]

We will test the Makefile under zcu102, but there have many problems when using zynq 7000 building platform, long for more supports.

Guoping Chen

imrickysu, 2022-08-01:
Hi @chenguoping76 , I didn't find `axi_ic_processing_system7_0_S_AXI_ACP` in your block design. Please check how you exported the AXI interfaced in Platform Setup. 

Please refer to [ZC706 base platform source code](https://github.com/Xilinx/Vitis_Embedded_Platform_Source/tree/2022.1/Xilinx_Official_Platforms/xilinx_zc706_base) for how to create custom platforms on ZYNQ-7000.

Since the tutorial work properly, I would classify this is not a tutorial issue. For additional design questions, please post your questions on http://forums.xilinx.com. AMD/Xilinx Technical Support will be glad to help. 

BTW, 2022.1 would be the last version that we support ZYNQ-7000 as acceleration platforms. From 2022.2, the ZC706 base platform will discontinue. There will be no testing on ZYNQ-7000 Vitis acceleration flow.

chenguoping76, 2022-08-02:
@imrickysu thanks for your help, we will submit problems to http://forums.xilinx.com/

closed 2022-08-02 by chenguoping76