Xilinx/Vitis-Tutorials #280: Vitis_Platform_Creation-kv260 step3 Load the hardware sudo xmutil command 'xmutil' not found

chenguoping76, 2022-08-02:
Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/step3.md 

4.  Load the hardware 
    sudo xmutil listapps
    command 'xmutil' not found

imrickysu, 2022-08-02:
Hi @chenguoping76 , xmutil is a special command only available in KV260 rootfs, which is prebuilt and write to SD card directly.
If you use the rootfs from common image, it doesn't include xmutil utility.

sven2314, 2022-08-05:
@chenguoping76 Please follow the following link to update the sdcard Image which is official image for kv260 and contains the xmutil command.
[kv260-getting-started](https://www.xilinx.com/products/som/kria/kv260-vision-starter-kit/kv260-getting-started/getting-started.html)

chenguoping76, 2022-08-06:
3ks，I got the difference. Kv260 has two prebuild system: ubuntu is a desktop; petalinux editor is hw acceleration platform.

closed 2022-08-06 by chenguoping76