Xilinx/Vitis-Tutorials #281: FPGA performance question

anonymous1782, 2022-08-02:
Hi, 

I'm new at FPGA and have an interest in Vitis.

First, I encountered performance bottleneck during FPGA write.

```
$ xbutil validate -d b3:00.1 -r 'dma'
Starting validation for 1 devices
 
Validate Device           : [0000:b3:00.1]
    Platform              : xilinx_u200_gen3x16_xdma_base_1
-------------------------------------------------------------------------------
Test 1 [0000:1a:00.1]     : DMA
    Details               : Host -> PCIe -> FPGA write bandwidth = 9302.8 MB/s
                            Host <- PCIe <- FPGA read bandwidth = 11909.4 MB/s
    Test Status           : [PASSED]
-------------------------------------------------------------------------------
```

Here is my "xbutil validate" result screen. It shows that FPGA write bandwidth is significantly less than FPGA read bandwidth.

[image]

However, according to the above graph (from [07-host-code-opt](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.1/Hardware_Acceleration/Design_Tutorials/07-host-code-opt), it shows about 30,000MB/s.

Here are my questions:
1. How can the above figure achieve larger than xbutil validate PCIe bandwidth? 
2. How can I improve "Host -> PCIe -> FPGA write bandwidth = 9302.8 MB/s" bandwidth? 

Thanks

imrickysu, 2022-08-02:
@Rampagee Could you share more info about the case with 30,000MB/s PCIe throughput? 

Hi @anonymous1782 , for general Vitis questions, we recommended you post in http://forums.xilinx.com/. AMD/Xilinx Tech Support team will be glad to support.

anonymous1782, 2022-08-02:
@imrickysu Thank you.

1. For 30,000MB/s PCie throughput, any information would be helpful.

2. I am looking for any related official documents about PCIe 3x16 xdma vitis performance. But I can not. Almost test shows write bandwidth << read bandwidth and it is not reasonable I think. Anyway, I will try to figure out more.

Again, I am looking forward to any information about the figure. Thank you.

anonymous1782, 2022-08-04:
In my understanding, the unit of this figure is not MBytes but MBit. I think it is quite confusing.
Further, I recognized that FPGA throughput is one direction throughput, and FPGA PCIe throughput includes read/write throughput according to the [host code](https://github.com/Xilinx/Vitis-Tutorials/blob/2022.1/Hardware_Acceleration/Design_Tutorials/07-host-code-opt/reference-files/src/pipeline_host.cpp#L79-L90).

Thanks.

Rampagee, 2022-08-29:
Sorry for late response, hi, @anonymous1782, I run the sweep test and get following figure. Sure, you are right that the Y axis is FPGA throughput and the unit is Mega Bits. @randyh62 , could you help to give more explanations here? 
[image]

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no further open concerns. 

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx