Xilinx/Vitis-Tutorials #283: Questions about host memory access

anonymous1782, 2022-08-04:
Hi, I have a question about [host memory access](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.1/Hardware_Acceleration/Feature_Tutorials/08-using-hostmem).

In common where Vitis xdma is sued, the xdma block design module in FPGA hardware is in charge of data transfer between host memory and FPGA global memory.

However, host memory access can be used no-DMA platform, and even it does not require xdma platform. Then, I wonder who is in charge of data transfer. 

Which component read/write host memory data for host memory access? 

Thanks

ravicho, 2022-08-04:
Hi @anonymous1782,

In the case of no-DMA platforms, an additional Bridge IP is used by the kernel(s) to read and write data directly from/to the Host Memory. This bridge IP provides DMA bypass capability which is primarily used for data transfer. 

Hope this answers your question.

Regards
Ravi

anonymous1782, 2022-08-05:
@ravicho Thank you for the detailed explanation. I see the point.

Then.. for DMA-enabled systems, are they using different IPs? In other words, is it correct that DMA for global memory uses xdma IP while host memory access uses bridge IP?

Further, are they interfered with by each other when they operate simultaneously? (DMA IP affects Bridge IP / Bridge IP affects DMA IP) 

Although I know my question is quite tricky and corner case, any talk would be helpful for me to understand host memory access.

Best regards,

AnusheelXilinx, 2025-01-13:
Closing the discussion as Never Fix. Generic XRT and host queries can be discussed at AMD Community Forums.

Thanks
Anusheel

closed 2025-01-13 by AnusheelXilinx