Xilinx/Vitis-Tutorials #284: Hello! How can I create a ZCU106 platform

HankIC, 2022-08-05:
Hello friends, I am a newcomer to Vitis-Ai. I have one ZCU106 board. How Can I create a platform for newest Vitis Ai?
Thank you!

chenguoping76, 2022-08-06:
ref https://github.com/Xilinx/Vitis-Tutorials/issues/278

imrickysu, 2022-08-10:
We don't have pre-built ZCU106 base platform, but ZCU106 is very similar to ZCU104. 

You can follow https://github.com/Xilinx/Vitis-Tutorials/tree/2022.1/Vitis_Platform_Creation/Design_Tutorials/02-Edge-AI-ZCU104 and use Vivado ZCU106 board file.

You can also use the ZCU104 base platform source code, replace the zcu104 board with zcu106.
https://github.com/Xilinx/Vitis_Embedded_Platform_Source/blob/2022.1/Xilinx_Official_Platforms/xilinx_zcu104_base/hw/xsa_scripts/project.tcl

closed 2022-08-19 by imrickysu