Xilinx/Vitis-Tutorials #285: hw acceleratate : future_tutorial, 02-mix c and rtl case Segmentation fault (core dumped)

chenguoping76, 2022-08-07:
[image]

and make xclbin command should be make xclbin DEVICE=<Platform>  ?

randyh62, 2022-08-10:
Hello @chenguoping76, Can you clarify what you are looking for here? The _make xclbin_ command can be used with _make xclbin DEVICE=XXX_ if you are specifying a new platform. 

As for the segmentation fault, as you are using a custom platform I don't know if I can help you with that.

chenguoping76, 2022-08-11:
yes @randyh62 I use custom platform for pynq z2 board, and test it with vadd hw accelator done. So I want to know how to use sw_emu/hw_emu evaluation for these zynq7000 embedded system. 3ks.  By the way, does this sw_emu evaluation can be verified on zc706 platform?

randyh62, 2022-08-11:
Hello @chenguoping76, you could try the tutorial with the zc706 platform and see how it works out. You can specify the device in the make command: 
**make run TARGET=sw_emu DEVICE=xilinx_zc706_base_202210_1**

These are my results:
```
XCL_EMULATION_MODE=sw_emu ./host krnl_vadd.sw_emu.xilinx_zc706_base_202210_1.xclbin
Using FPGA binary file specfied through the command line: krnl_vadd.sw_emu.xilinx_zc706_base_202210_1.xclbin
Found Platform
Platform Name: Xilinx
Loading: 'krnl_vadd.sw_emu.xilinx_zc706_base_202210_1.xclbin'
Kernel Name: krnl_vadd_1, CU Number: 0, Thread creation status: success
TEST WITH ONE KERNEL PASSED
device process sw_emu_device done
```

randyh62, 2022-08-11:
I would suggest you work with a standard platform to understand the tutorial and the tools, and then work with your customized platform to see where it varies. You can get additional support on the user forum.

randyh62, 2022-08-15:
As this is an issue with the custom platform, and not the tutorial, I am closing this issue.

closed 2022-08-15 by randyh62

chenguoping76, 2022-08-16:
3ks