Xilinx/Vitis-Tutorials #287: Could not create zynqmp_fsbl application for the given hw design | Details: FSBL uses these IPs but are missing in the design: psu_i2c_0

HankIC, 2022-08-11:
Hello! I follow the instructions in "https://github.com/Xilinx/Vitis-Tutorials/tree/2021.2/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104" and change some board information to ZCU106 in step1 and step2. In the end, this problem happens, how can I fix this problem in makefile or tcl?
[error_in_terminal.txt](https://github.com/Xilinx/Vitis-Tutorials/files/9310266/error_in_terminal.txt)

sven2314, 2022-08-19:
Hi @HankIC,
I checked the zcu106 board's schematic. The I2C0 is used to do power related control. So enabling the I2C0 in Vivado design is required.
Please add I2C0 support in system-step1.tcl file like the following 'set property' line is the code. Other lines are to point the location.
```
# Create address segments
  assign_bd_address -offset 0x80000000 -range 0x00010000 -target_address_space [get_bd_addr_spaces zynq_ultra_ps_e_0/Data] [get_bd_addr_segs axi_intc_0/S_AXI/Reg] -force
  set_property -dict [list CONFIG.PSU__I2C0__PERIPHERAL__ENABLE {1} CONFIG.PSU__I2C0__PERIPHERAL__IO {MIO 14 .. 15}] [get_bd_cells zynq_ultra_ps_e_0]
  # Restore current instance
  current_bd_instance $oldCurInst
```
You can also enable the I2C0 support in Vivado GUI. Just double click the Zynq UltraScale+ MPSoC IP and go to IO configuration, then enable the I2C0 and set the PIN group to MIO 14, 15.
After that  you will not encounter that issue.

HankIC, 2022-08-19:
> Have you tried on a newer version of Vitis? for example 2022.1

Hello @allyzhou , its difficult for me to change a new version because VITIS installer is so big and Ubuntu network is not good enough. It may cost me several days to download a installer from website to windows PC and upload the installer to baiduyuncloud and download the installer from baiduyuncloud to ubuntu T-T

HankIC, 2022-08-19:
Hello @sven2314 , thank you for your advise! I haven't got new result up to now because I have run step2 many times this evening. Step2 always get stuck in not fixed point, the program will stop for a long time and no new progress. So I have to kill the process and run step2 again. Now step2 stops again! it has been stuck here for 26 minutes!
[image]

HankIC, 2022-08-22:
Hello @sven2314, thank you!  the problem disappeared!!!!
@allyzhou @sven2314 thank you very much!

closed 2022-08-22 by imrickysu