Xilinx/Vitis-Tutorials #292: Using 400 AIE tiles through PLIO problem

CoffeeCat3008871, 2022-08-24:
I am trying to utilize all 400 AIE tiles to process the simple adder graph kernel. However, I encountered a problem shown at the bottom. For the adder kernel, it requires 2 input stream and 1 output stream. I therefore created 400 PLIO for input1 and input 2, and 400 output. There are 1200 PLIO created in total. However, it seems that the maximum number of IO I can create is 900. But the error message shows that using -ftemplate-depth can help but I had tried all place in makefile I could but nothing works so far. Reducing the number of PLIO to 100 for each with a total of 300 PLIO works well. Can anyone help with this problem?

src/aie/adder_x400.cpp:2414:48:   required from here
/tools/Xilinx/Vitis/2022.1/aietools/include/adf/new_frontend/adf_api_impl.h:120:31: fatal error: template instantiation depth exceeds maximum of 900 (use -ftemplate-depth= to increase the maximum)
             getPlatformIoAttrs(ioAttrOrFiles, ios...);
             ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ERROR: [aiecompiler 77-753] This application has discovered an exceptional condition from which it cannot recover while executing the following command
  >> ${XILINX_VITIS_AIETOOLS}/tps/lnx64/gcc/bin/g++  -std=c++17  -I .  ./Work/temp/adder_x400.processed.ii -o ./Work/temp/adder_x400.out -L /tools/Xilinx/Vitis/2022.1/aietools/lib/lnx64.o -g -O0 -Wl,--unresolved-symbols=ignore-all  -Wno-return-stack-address  -Wno-missing-declarations  -lmeir_frontend  -ladf_api_frontend .
Please check the output log for errors and fix those before you run the application.
(WARNING:0, CRITICAL-WARNING:0, ERROR:1)
/tools/Xilinx/Vitis/2022.1/aietools/bin/aiecompiler: line 83: kill: (-22912) - No such process
makefile_versal_ps.mk:141: recipe for target 'libadf.a' failed
make: *** [libadf.a] Error 255

OTremois, 2022-08-29:
Hi, I am Olivier from AMD AECG. Can you share the tutorial you used?
Thanks
Olivier

CoffeeCat3008871, 2022-08-29:
> Hi, I am Olivier from AMD AECG. Can you share the tutorial you used? Thanks Olivier

Hi, Olivier, thank you for your help. I used this tutorial as a base framework.
https://github.com/Xilinx/Vitis-Tutorials/tree/2022.1/AI_Engine_Development/Design_Tutorials/08-n-body-simulator

The difference is that I did not use the technique of Explicit Packet Switching since in the simple adder example, there is only two input stream.

CoffeeCat3008871, 2022-08-29:
That's very helpful. Thank you a lot!

closed 2022-08-29 by CoffeeCat3008871

jamstine, 2022-12-05:
does anyone have a packet switching example to look at to address using all the 400 AI engines?

Albresky, 2024-11-02:
> If you do not use packet switching mechanism then you are limited to 312 input streams (39 available columns x 8 streams (64 bits) per column, which is further re-arranged inside the AI Engine array into 50 available columns x 6 streams(300)), and 234 outputs (39 columns x 6 streams per column, coming from 50 columns x 4 streams (200) inside the array). So you can implement a maximum of 150 simple adders (I would say a little less as routing to the PL may be difficult) without using packet switching mechanism. In order to use the full 400 AI Engines you nee to use packet switching to share inputs and output among the 4 kernels of the nbodySubsystem graph.

---
Hi, Olivier, I've encountered a PLIO related issues as well. 

As known to all, there are 39 PLIO ports available. In detail, in 64bit data width, 8 channels are available for each PLIO port from PL to AIEs and 6 channels from AIEs to PL. So, when the datawidth expands from 64bit to 128bit, there should be $64/128 \times 8 \times 39$=156 PLIO channels in total from PL to AIEs.

When I use aiecompiler in Vitis 2021.1 to compile this graph, a graph with 88 input PLIOs and 33 ouput PLIOs both in 128bit could be compiled successfully. But, when I used aiecompiler in Vitis 2023.2 and later(so far, Vitis 2024.1), the aiecompiler failed with *"ERROR: [AIE-PRE-MAPPER-11] area group ((0, 0), (49, 0)) has capacity 78 PLIO 128 bit incoming registered channels"*.

Is this a tool bug? I've seen several users encounted this issue but no response was given from AMD/Xilinx official.

Could you help to explain how PLIOs works under different datawidth? Many thanks.

kmhatre14, 2025-02-09:
@Albresky Did you find any solution to this problem?

OTremois, 2025-02-10:
Hi, this might be due to the PL clock that could be too high:

Each internal stream can be connected to :

- 64b PL stream running @ PL Clk ≤ AIE Clk/2
- 32b PL stream running @ PL Clk ≤ AIE Clk/2 using half width of the 64b PL stream
- 128b PL stream combining 2 contiguous 64b PL streams running @ PL Clk ≤ AIE Clk/4

- All south PL streams can be registered
- 2 north PL streams (stream 3 & 7) can’t be registered, 6 others can be registered.
	- In case of 128b PL the whole interface i/f will be unregistered