Xilinx/Vitis-Tutorials #296: No more output from terminal(ZCU106)

HankIC, 2022-08-30:
Hello, I follow the tutorials (https://github.com/Xilinx/Vitis-Tutorials/tree/2021.2/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104) and change board information from ZCU104 to ZCU106 in step1 and step2 and step3. However, when I test the platform in step4, I cannot get correct output and don't know how to debug this problem. Could any friends can give me some advice？ Thank you very much!
[image]
[image]
[scripts.zip](https://github.com/Xilinx/Vitis-Tutorials/files/9453851/scripts.zip)

HankIC, 2022-08-31:
New update: I use official **ZCU104** [image](https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-zcu104-dpu-v2021.1-v1.4.0.img.gz) and boot it in **ZCU106**. I get the same output. Maybe some configuration other than boardname should be changed in scripts.
[image]

HankIC, 2022-09-01:
In the step2, I find zcu106 dts is different from zcu104 according to [link](https://lore.kernel.org/all/CAHTX3dKFBiGpkK0XcWacArxA7KGxrYj_L1BpFyRzwmGP36e-sA@mail.gmail.com/T/#Z2e.:..:e29ef27a6a3bd52784b5e59f20d8b7f9b02a8e9d.1578396789.git.michal.simek::40xilinx.com:1arch:arm:dts:zynqmp-zcu106-revA.dts). I try to change it and update the result later

HankIC, 2022-09-02:
New update: Vadd example can run, but DPU example can't work and ZCU106 cannot boot from SD card. I don't know how to fix this issue now.
[image]

HankIC, 2022-09-02:
New update: It can't work at the first time, but when I try to boot from SD for the third time, It can work!!!!!
I wonder why my project is so unstable?????

sven2314, 2022-11-22:
@HankIC Hi Hank, 
Please check  this link: https://github.com/Xilinx/Vitis-Tutorials/issues/287 to update the TCL file to support zcu106 and enable the I2C0 support.
When you build the PeTalinux project in step2 please change the machine name to "zcu106-reva". Then it will automatically add the zcu106 board level dtsi file. That means you will not need to modify the DTS file manually.

AnusheelXilinx, 2024-12-21:
Resolution was provided in the last post. Closing as there are no further queries.

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx