Xilinx/Vitis-Tutorials #299: HwAcc/Intro: no matching function for call to 'resize'

PDZZXL, 2022-09-08:
I move the "Vitis_Libraries" to "Hardware_Acceleration/Introduction/design_source/hw_src/" and executed the "make" command. Then the error occurs：
ERROR: [v++ 207-3334] no matching function for call to 'resize' (/home/foxken_zxl/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_src/resize_rgb.cpp:77:9)

I did not change any code，so how do I solve my problem?
Thanks！

imrickysu, 2022-09-19:
Hi @PDZZXL , it's an issue in the makefile. Please run `git clone https://github.com/Xilinx/Vitis_Libraries` in `Hardware_Acceleration/Introduction/design_source/hw_src/` directory. We will fix this issue. Thanks.

AnusheelXilinx, 2022-11-21:
This is a known issue due to changes in the latest Vitis_Libraries. This issue will be fixed in 2022.2 branches soon. 
For now, please use the older version of Vitis_Libraries(https://github.com/Xilinx/Vitis_Libraries/tree/2021.1)

AnusheelXilinx, 2022-11-21:
We have now published the latest version of this tutorial. You can refer below link for the updated tutorial:
https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/Hardware_Acceleration/Design_Tutorials/10-get_moving_with_alveo

AnusheelXilinx, 2024-12-21:
Fix was updated as per the last post. Closing the issue as there are no further queries.

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx