Xilinx/Vitis-Tutorials #304: Vitis-Tutorials/AI_Engine_Development/Design_Tutorials/08-n-body-simulator iteration problem

CoffeeCat3008871, 2022-09-27:
I am working on adding more iterations to this tutorial. however I have been facing design timing problem when more iterations are added to PL kernels. For example, I added a " for (int iter=0; iter<NUMBER_ITER; iter++)" to Line 239, in file "Module_03_pl_kernels/kernel/packet_sender.cpp" and kept the rest to remain the same. The loop structure in the file looks as followings: 
 for (int iter=0; iter<NUMBER_ITER; iter++) {
  for(int cu=0; cu<NUM_CU; cu++){
    for (int h = 0; h < PACKET_NUM; h++) 
      { 
	unsigned int ID=packet_ids[h];
	ap_uint<32> header=generateHeader(PKTTYPE,ID); //packet header
	tmp.data=header;
	tmp.keep=-1;
	tmp.last=0;
...
...

I added exactly the same iteration loop to hls_packet_receiver.cpp, m2s, and s2m files. However, this design was reported with a design timing error and failed to build .xsa file. When II was set to 2 in all files, the whole design could be built but could only run below 3 iterations. (Above will be stall). Can anyone provide some suggestion about how to make the entire design to be run for multiple iterations without a problem? (I also tried to use UNROLL and other pipeline style but none of them seemed to be working)

xflorentw, 2022-09-28:
HI @CoffeeCat3008871 

This is outside of the scope of the tutorial so the question is not really a good fit for the github issue.

I would recommend that you post on the Xilinx forums with the HLS tag as this seems to be related to HLS
https://support.xilinx.com/s/topic/0TO2E000000YKXtWAO/hls

CoffeeCat3008871, 2022-09-29:
Sure, thanks.

closed 2022-09-29 by CoffeeCat3008871