Xilinx/Vitis-Tutorials #310: createdts fails for KV260 XSA

shareefj, 2022-10-14:
Hi,

I'm following the KV260 tutorial and am seeing the following error when trying to generate the device tree.  I'm running 2022.1 on Ubuntu 22.04.  What am I  doing wrong here?  I'm following https://github.com/Xilinx/Vitis-Tutorials/blob/2022.1/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/step2.md.

```
setws .
createdts -hw ../kv260_hardware_platform/top_fpga_kv260.xsa -zocl -platform-name genesee -git-branch xlnx_rel_v2022.1 -overlay -compile
INFO: Downloading DTG repo from https://github.com/Xilinx/device-tree-xlnx.git to /home/shareefj/git/genesee_fpga/workspace/kv260_vitis_platform
Cloning into 'device-tree-xlnx'...
remote: Enumerating objects: 12008, done.
remote: Counting objects: 100% (3376/3376), done.
remote: Compressing objects: 100% (747/747), done.
remote: Total 12008 (delta 2141), reused 3350 (delta 2118), pack-reused 8632
Receiving objects: 100% (12008/12008), 2.25 MiB | 8.33 MiB/s, done.
Resolving deltas: 100% (6478/6478), done.
INFO: [Hsi 55-2053] elapsed time for repository (/opt/Xilinx/Vitis/2022.1/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 2414.035 ; gain = 0.000 ; free physical = 1302 ; free virtual = 35887
INFO: Creating platform genesee at workspace /home/shareefj/git/genesee_fpga/workspace/kv260_vitis_platform
Opening the hardware design, this may take few seconds.
Starting vitis. This could take few seconds...Eclipse:
The Eclipse executable launcher no longer supports running with GTK + 2.x. Continuing using GTK+ 3.x.
done
INFO: Populating the default qemu data for the domain "device_tree_domain" from the install location /opt/Xilinx/Vitis/2022.1/data/emulation/platforms/zynqmp/sw/a53_standalone/qemu/
zocl:false
ext_platform:
WARNING: psu_usb_xhci_0: No reset found
WARNING: label 'usb0' found in existing tree
INFO: Generating device tree
zocl:true
ext_platform:
Failed to find overlay2 node !!!
ERROR: [Hsi 55-1545] Problem running tcl command ::sw_device_tree::generate : Failed to find overlay2 node !!!
    while executing
"error "Failed to find $lu_node node !!!""
    (procedure "get_node_object" line 31)
    invoked from within
"get_node_object overlay2 pl.dtsi"
    invoked from within
"create_dt_node  -name zyxclmm_drm -objects [get_node_object overlay2 pl.dtsi]"
    ("eval" body line 1)
    invoked from within
"eval "create_dt_node ${cmd}""
    (procedure "add_or_get_dt_node" line 187)
    invoked from within
"add_or_get_dt_node -n "zyxclmm_drm" -d ${default_dts} -p $bus_node"
    (procedure "gen_zocl_node" line 21)
    invoked from within
"gen_zocl_node"
    (procedure "::sw_device_tree::generate" line 23)
    invoked from within
"::sw_device_tree::generate device_tree"
ERROR: [Hsi 55-1442] Error(s) while running TCL procedure generate()
Failed to generate the bsp sources for domain.device_tree_domain
```

shareefj, 2022-10-14:
I switched to the xlnx_rel_v2022.1 branch and this succeeds.

```
createdts -hw ../kv260_hardware_platform/top_fpga_kv260.xsa -zocl -platform-name genesee -git-branch xlnx_rel_v2022.2 -overlay -compile
INFO: Downloading DTG repo from https://github.com/Xilinx/device-tree-xlnx.git to /home/shareefj/git/genesee_fpga/workspace/kv260_vitis_platform
Cloning into 'device-tree-xlnx'...
remote: Enumerating objects: 12008, done.
remote: Counting objects: 100% (3376/3376), done.
remote: Compressing objects: 100% (747/747), done.
remote: Total 12008 (delta 2141), reused 3350 (delta 2118), pack-reused 8632
Receiving objects: 100% (12008/12008), 2.25 MiB | 2.96 MiB/s, done.
Resolving deltas: 100% (6478/6478), done.
INFO: [Hsi 55-2053] elapsed time for repository (/opt/Xilinx/Vitis/2022.1/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 2390.285 ; gain = 0.000 ; free physical = 2113 ; free virtual = 37432
INFO: Creating platform genesee at workspace /home/shareefj/git/genesee_fpga/workspace/kv260_vitis_platform
Opening the hardware design, this may take few seconds.
Starting vitis. This could take few seconds...Eclipse:
The Eclipse executable launcher no longer supports running with GTK + 2.x. Continuing using GTK+ 3.x.
done
INFO: Populating the default qemu data for the domain "device_tree_domain" from the install location /opt/Xilinx/Vitis/2022.1/data/emulation/platforms/zynqmp/sw/a53_standalone/qemu/
zocl:true
ext_platform:
WARNING: dt_zocl enabled and No PL ip's found in specified design, skip adding zocl node
WARNING: psu_usb_xhci_0: No reset found
WARNING: label 'usb0' found in existing tree
INFO: Generating device tree
zocl:true
ext_platform:
WARNING: dt_zocl enabled and No PL ip's found in specified design, skip adding zocl node
WARNING: psu_usb_xhci_0: No reset found
WARNING: label 'usb0' found in existing tree
Building the BSP Library for domain  - device_tree_domain on processor psu_cortexa53_0
make: Nothing to be done for 'all'.

system-top.dts:24.9-27.4: Warning (unit_address_vs_reg): /memory: node has a reg or ranges property, but no unit name
zynqmp.dtsi:789.43-793.6: Warning (pci_device_reg): /axi/pcie@fd0e0000/legacy-interrupt-controller: missing PCI reg property
```

shareefj, 2022-10-14:
I'm guessing the relevant fix is the following where the script returns if the PL has no IPs.
```
git diff -r 14cc0faa539470dea24b196d278742b118a46b6d -r 0124aad7d9f0e50062aa79189c80f183956987e2 device_tree/data/device_tree.tcl
<snip>
@@ -701,6 +741,18 @@ proc gen_zocl_node {} {
        if {!$zocl} {
                return
        }
+       #Check if design has any PL ip's
+       set ip_count 0
+       foreach ip [get_drivers] {
+               if {[is_pl_ip $ip]} {
+                       incr ip_count
+                       break
+               }
+       }
+       if {$ip_count == 0} {
+               dtg_warning "dt_zocl enabled and No PL ip's found in specified design, skip adding zocl node"
+               return
+       }
        set dt_overlay [get_property CONFIG.dt_overlay [get_os]]
        if {$dt_overlay} {
                set bus_node "overlay2"
```

sven2314, 2022-11-28:
@shareefj Hi, I tried it in my side. I can not reproduce this issue. If you could provide your project I would help to investigate the  root cause.

AnusheelXilinx, 2024-12-21:
Issue not reproducible. Closing the issue. 

thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx

chim20air, 2025-04-11:
Hi, I also get this error, but using the [2024.2 version](https://github.com/Xilinx/Vitis-Tutorials/blob/2024.2/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260/step2.md). I don't have access to the log file, but yes to the window outputs

[image]

[image]

sven2314, 2025-04-14:
hi @chim20air could you provide you XSA file for further investigation?