Xilinx/Vitis-Tutorials #311: Vector Addition example doesn't build

shareefj, 2022-10-20:
I'm trying to run the KV260 platform creation tutorial and am seeing an error in step 3 when building the example vadd application.  At the point where we build the vadd_system_hw_link.prj, and from what I've debugged so far, the `vpl.tcl` script falls over when trying to rebuild the vivado project.

I don't really understand what the scripts are trying to do here but it looks like the `create_project` step fails because it's looking in the wrong path.  I've copied the log file below and the directory structure under `kv260_vitis_platform/vadd_system_hw_link/Hardware/binary_container_1.build/link/vivado` looks like:

```
.
└── vpl
    ├── output
    ├── prj
    │   ├── prj.cache
    │   ├── prj.gen
    │   ├── prj.hw
    │   ├── prj.ip_user_files
    │   └── prj.srcs
    └── scripts
```

The error below is due to the fact that the srcs path is incorrect: `kv260_vitis_platform/vadd_system_hw_link/Hardware/binary_container_1.build/link/vivado/vpl/prj/kv260_hardware_platform.srcs/utils_1/imports/synth_1/system_wrapper.dcp`

```
#-----------------------------------------------------------
# Vivado v2022.1.2 (64-bit)
# SW Build 3605665 on Fri Aug  5 22:52:02 MDT 2022
# IP Build 3603185 on Sat Aug  6 04:07:44 MDT 2022
# Start of session at: Fri Oct 21 00:30:42 2022
# Process ID: 20715
# Current directory: /home/shareefj/git/genesee_fpga/workspace/kv260_tutorial/kv260_vitis_platform/vadd_system_hw_link/Hardware/binary_container_1.build/link/vivado/vpl
# Command line: vivado -log manual.log -mode batch -source vpl.tcl -notrace
# Log file: /home/shareefj/git/genesee_fpga/workspace/kv260_tutorial/kv260_vitis_platform/vadd_system_hw_link/Hardware/binary_container_1.build/link/vivado/vpl/manual.log
# Journal file: /home/shareefj/git/genesee_fpga/workspace/kv260_tutorial/kv260_vitis_platform/vadd_system_hw_link/Hardware/binary_container_1.build/link/vivado/vpl/vivado.jou
# Running On: JALAPINO, OS: Linux, CPU Frequency: 3800.260 MHz, CPU Physical cores: 24, Host memory: 33627 MB
#-----------------------------------------------------------
source vpl.tcl -notrace
[00:30:45] Run vpl: Step create_project: Started
INFO: [OCL_UTIL] current step: vpl.create_project
Creating Vivado project.
INFO: [OCL_UTIL] set ::origin_dir_loc .local/hw_platform/prj
INFO: [OCL_UTIL] set ::user_project_name prj
INFO: [OCL_UTIL] internal step: source .local/hw_platform/prj/rebuild.tcl to create prj project
INFO: [Project 1-1727] Source BDs should be added before importing the top BD /home/shareefj/git/genesee_fpga/workspace/kv260_tutorial/kv260_vitis_platform/vadd_system_hw_link/Hardware/binary_container_1.build/link/vivado/vpl/.local/hw_platform/prj/kv260_hardware_platform.srcs/sources_1/bd/system/system.bd with BDC
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2022.1/data/ip'.
ERROR: caught error: ERROR: [Common 17-69] Command failed: File '/home/shareefj/git/genesee_fpga/workspace/kv260_tutorial/kv260_vitis_platform/vadd_system_hw_link/Hardware/binary_container_1.build/link/vivado/vpl/prj/kv260_hardware_platform.srcs/utils_1/imports/synth_1/system_wrapper.dcp' does not exist

    while executing
"rdi::set_property -name incremental_checkpoint -value /home/shareefj/git/genesee_fpga/workspace/kv260_tutorial/kv260_vitis_platform/vadd_system_hw_lin..."
    invoked from within
"set_property -name "incremental_checkpoint" -value "$proj_dir/kv260_hardware_platform.srcs/utils_1/imports/synth_1/system_wrapper.dcp" -objects $obj"
    (file ".local/hw_platform/prj/rebuild.tcl" line 272)
    invoked from within
"source $hw_platform_rebuild_tcl"
[00:30:50] Run vpl: Step create_project: Failed
INFO: [OCL_UTIL] current step: vpl.create_project failed. To rerun the existing project please use --from_step vpl.create_project
Failed to rebuild a project required for hardware synthesis. The project is 'prj'. The rebuild script is '.local/hw_platform/prj/rebuild.tcl'. The rebuild script was delivered as part of the hardware platform. Consult with the hardware platform provider to investigate the rebuild script contents. An error stack with function names and arguments may be available in the 'vivado.log'.
INFO: [Common 17-206] Exiting Vivado at Fri Oct 21 00:30:50 2022...
```

Should this tutorial be working with 2022.1.2?

JH989876525, 2022-10-24:
I get into the similar problem on DPU-TRD with 2022.1 too.

I find out that it was caused by the **rebuild.tcl** in .xsa, it  will do some check that may be unnecessary.

Generate the .xsa by following flow could get the **rebuild.tcl** without these check.
Success work flow:
Create vivado block design -> Generate Block Design -> Exprot platform to generate .xsa

Failed work flow:
Create vivado block design -> Generate Bitstream -> Exprot platform to generate .xsa

Hope this post could help you out.

JH

shareefj, 2022-10-24:
@JH989876525 Thanks Jerry, that seems to have fixed the issue.

So anyone from Xilinx, can you update your instructions or fix this issue if it shouldn't happen?  The offending parts of the instructions are:

```
Generate the design

Select Generate Block Design from Flow Navigator
Select Synthesis Options to Global. It will skip IP synthesis during generation.
Click Generate.
Click Generate Bitstream in Flow Navigator window.
Click Yes.
Export the platform

Click menu File -> Export -> Export Platform to launch the Export Hardware Platform wizard. This wizard can also be launched by Export Platform button in Flow Navigator or Platform Setup window.
Click Next in the first information page.
Select Platform Type: Hardware and Hardware Emulation, click Next. If you skipped the emulation setup previously, select Hardware here.
Select Platform State: Pre-synthesis, enable Include Bitstream, click Next
Input Platform Properties and click Next. For example,
Name: kv260_hardware_platform
Vendor: xilinx
Board: kv260
Version: 0.0
Description: This platform provides high PS DDR bandwidth and three clocks: 100MHz, 200MHz and 400MHz.
Fill in XSA file name: kv260_hardware_platform and keep the export directory as default.
Click Finish.
kv260_hardware_platform.xsa will be generated. The export path is reported in the Tcl console.
```

JH989876525, 2023-08-17:
Hi all,

After 10 month I comes into this issue again on VCK190 with Vitis AI 3.0 DPU-TRD flow.

This time I need the pdi file(bitstream) within the xsa, so I can not using the solution which I used on k26.

I find out the solution of this issue once and for all with following xilinx forum.
https://support.xilinx.com/s/question/0D52E000073xiQCSAY/vitis-cannot-find-the-dcp-file-it-extracted-from-the-xsa?language=en_US

The dcp file should be not in xsa file, so delete it befroe genrating xsa file.
You do it on Vivado GUI Source -> Utility Sources, or you can add some tcl cmd like this in the tcl file.
```tcl
export_ip_user_files -of_objects  [get_files $proj_dir/${proj_name}.srcs/utils_1/imports/synth_1/${proj_name}_wrapper.dcp] -no_script -reset -force -quiet
remove_files  -fileset utils_1 $proj_dir/${proj_name}.srcs/utils_1/imports/synth_1/${proj_name}_wrapper.dcp
```

BR,
JH

AnusheelXilinx, 2024-12-21:
Hi @sven2314, can you verify if the changes were added. Closing the issue as there are not further open queries. 

thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx