Xilinx/Vitis-Tutorials #313: The document you are looking for has been moved or deleted

zyt1024, 2022-11-04:
when I tried to click  ”For more information, refer to [Enabling the Vitis Kernel Flow](https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/Vitis-HLS-Process-Overview) in the Vitis HLS Flow of the Vitis Unified Software Platform Documentation (UG1416)“. the link will tell me this message `The document you are looking for has been moved or deleted`.[
[image]
](url)

imrickysu, 2022-11-04:
Hi @zyt1024 , pleas visit https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/Vitis-HLS-Flow-Overview.
Thanks for reporting this issue. We will update the doc accordingly.

imrickysu, 2022-11-04:
Hi @randyh62 , there are many places referred to this chapter. Could you help to confirm and update? Thanks.
https://github.com/Xilinx/Vitis-Tutorials/search?q=Enabling+the+Vitis+Kernel+Flow

zyt1024, 2022-11-04:
> Hi @zyt1024 , pleas visit https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/Vitis-HLS-Flow-Overview. Thanks for reporting this issue. We will update the doc accordingly.
Thanks

randyh62, 2022-11-04:
I will fix this. Thanks.

randyh62, 2022-11-07:
This has been promoted and should be fixed in a few days. I will mark this issue closed. Thanks for bringing it to our attention.

closed 2022-11-07 by randyh62