Xilinx/Vitis-Tutorials #318: HwAcc/F/09-Ethernet-Alveo x4 on U200

eerobert, 2022-11-15:
Hi

I was trying to go thru the 09-Ethernet-Alveo using LANE=4

However, vitis was not able to complete the implementation and leaving an error of below

[Constraints 18-1000] Routing results verification failed due to partially-conflicted nets (Up to first 10 of violated nets):  level0_i/ulp/eth0/inst/i_ethernet_0/inst/i_core_gtwiz_userclk_tx_inst_1/gen_gtwiz_userclk_tx_main.bufg_gt_usrclk_inst_0 level0_i/ulp/memory_subsystem/inst/memory/ddr4_mem02/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/xlnx_opt_ 

Would like to know how to solve the above errors?

Thanks

imrickysu, 2022-11-17:
Hi @Rampagee , could you help with this issue?

Rampagee, 2022-11-17:
Hi, @eerobert, are you using all the default configuration with "make all LANE=4" command to build? Could you provide the platform and Vitis version, as well as the GitHub repo branch you are using (such as 2022.2, 2022.1, etc.)? Thanks.

eerobert, 2022-11-17:
i edit the makefile and change LANE := 1 to LANE := 4
and i am checking out 2022.1 branch and compiled with vivado 2022.1

Rampagee, 2022-11-18:
Hi, @eerobert, I tried and re-product the issue, I will dig into this and please use 2022.2 version temporarily. Thanks.

AnusheelXilinx, 2024-12-22:
With 2024.2, I was able to build the example with LANE := 4. Looks like the issue in discussion got fixed. 

Thanks
Anusheel

closed 2024-12-22 by AnusheelXilinx