Xilinx/Vitis-Tutorials #321: Multipliers in separate files

vrrao01, 2022-11-17:
I'm currently using Vitis HLS 2022.1. Whenever my source C code contains a multiplication operation, the multipliers are created as a separate Verilog module in a new file. Is there any way to disable this behaviour? I don't mind any performance impacts due to lack of pipelining or any other optimizations. 

**Source C code**
```
void hls_macc(signed int a, signed int b, int *max, bool accum_clr)
{
   static acc_reg = 0;
   acc_reg = a * b;
   *max = acc_reg;
}
```

The multiplication is realised in a `hls_macc_mul_32s_32s_32_2_1` module. I would like the multiplication to be in `hls_macc.v` itself.

imrickysu, 2022-11-21:
Hi @vrrao01 , it looks like your question is a generic issue, that is not related to a specific Vitis tutorial. Could you please ask this question in Xilinx forums? https://forums.xilinx.com/ 
It has an [HLS topic](https://support.xilinx.com/s/topic/0TO2E000000YKXtWAO/hls?language=en_US). Vitis HLS engineers will help you there.

imrickysu, 2022-11-21:
Hi @randyh62 , if you in case know that HLS has some configurations can change this behavior, please feel free to comment.

randyh62, 2022-11-21:
You could try the ALLOCATION pragma:
https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/pragma-HLS-allocation

closed 2022-12-16 by imrickysu