Xilinx/Vitis-Tutorials #325: Getting_Started/Vitis/Part4-data_center  software emulation fail

2748669816, 2022-11-24:
When I followed the tutorial to the fourth part of the software simulation, I failed。

The error returned looks like this--
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to open file './vadd.xclbin' for reading

I didn't find a solution.

randyh62, 2022-11-30:
The vadd.xclbin is created during the link process:
v++ -l -t sw_emu --platform xilinx_u200_gen3x16_xdma_2_202110_1 --config ../../src/u200.cfg ./vadd.xo **-o ./vadd.xclbin**

When you run this command does the tool return any errors?

randyh62, 2022-12-06:
As I have not heard back in a week I am closing the issue again. Feel free to reopen if necessary.