Xilinx/Vitis-Tutorials #331: ethernet-on-alveo x4 on u55c

roger90810, 2022-12-03:
Hello,

I'm trying to use ethernet kernel on Alveo u55c.
In the ethernet IP generate script `gen_ip_x4.tcl`, it doesn't support u55c.
I copy the property from u50 in the script to generate IP, but I failed at the build_hw step.
I want to ask whether this kernel does not support u55c or do I need to do other settings?
Thanks

```
****** vpl v2022.1 (64-bit)
  **** SW Build 3524075 on 2022-04-13-17:42:45
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/home/roger/Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/09-using-ethernet-on-alveo/_x/link/int/kernel_info.dat'.
INFO: [VPL 74-78] Compiler Version string: 2022.1
INFO: [VPL 60-423]   Target device: xilinx_u55c_gen3x16_xdma_3_202210_1
INFO: [VPL 60-1032] Extracting hardware platform to /home/roger/Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/09-using-ethernet-on-alveo/_x/link/vivado/vpl/.local/hw_platform
[22:36:02] Run vpl: Step create_project: Started
Creating Vivado project.
[22:36:04] Run vpl: Step create_project: Completed
[22:36:04] Run vpl: Step create_bd: Started
[22:36:11] Run vpl: Step create_bd: Failed
[22:36:11] Run vpl: FINISHED. Run Status: create_bd ERROR

===>The following messages were generated while  creating FPGA bitstream. Log file: /home/roger/Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/09-using-ethernet-on-alveo/_x/link/vivado/vpl/runme.log :
ERROR: [VPL 19-3476] Tcl error in create_gui procedure for BD Cell 'ulp_ulp_ucs_0/bd_22c0/bd_22c0_clkwiz_hbm_0'. expected floating-point number but got "Unable to get value from speedsfile for keyword MM"
ERROR: [VPL 19-3428] Failed to create Customization object ulp_ulp_ucs_0/bd_22c0/bd_22c0_clkwiz_hbm_0
ERROR: [VPL 41-1712] Create IP failed with errors
ERROR: [VPL 5-7] Error: running create_bd_cell  -vlnv xilinx.com:ip:clk_wiz:6.0 -type ip -name clkwiz_hbm .
ERROR: [VPL 7-5] XIT evaluation error: ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

ERROR: [VPL 19-167] Failed to deliver one or more file(s).
ERROR: [VPL 19-3541] IP Elaboration error: Failed to generate IP 'ulp_ucs'. Failed to generate 'Elaborate BD' outputs: Failed to elaborate IP.
ERROR: [VPL 41-1689] Failed to generate IP 'ulp_ucs'. Failed to generate 'Elaborate BD' outputs: Failed to elaborate IP.
ERROR: [VPL 41-595] Failed to add ip repository block <ulp_ucs>
ERROR: [VPL 41-425] Failed to read Diagram <ulp> from block design file </home/roger/Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/09-using-ethernet-on-alveo/_x/link/vivado/vpl/prj/prj.srcs/my_rm/bd/ulp/ulp.bd>
ERROR: [VPL 60-773] In '/home/roger/Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/09-using-ethernet-on-alveo/_x/link/vivado/vpl/runme.log', caught Tcl error:  ERROR: [Common 17-39] 'open_bd_design' failed due to earlier errors.
ERROR: [VPL 60-704] Integration error, Step failed: create_bd An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
WARNING: [VPL 60-1142] Unable to read data from '/home/roger/Vitis-Tutorials/Hardware_Acceleration/Feature_Tutorials/09-using-ethernet-on-alveo/_x/link/vivado/vpl/output/generated_reports.log', generated reports will not be copied.
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [22:36:11] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:05 ; elapsed = 00:00:16 . Memory (MB): peak = 2210.684 ; gain = 0.000 ; free physical = 23575 ; free virtual = 30204
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.

make: *** [Makefile:66: build_hw] Error 1
```

Rampagee, 2022-12-07:
Hi, @roger90810 , for U55C, please as below modifications:

1. change PLATFORM and DEVICE in Makefile:
~~~
PLATFORM := xilinx_u55c_gen3x16_xdma_3_202210_1
PART     := xcu55c-fsvh2892-2L-e
~~~

2.  use following lines to generate IP in gen_ip_x4.tcl
~~~
create_ip -name xxv_ethernet \
          -vendor xilinx.com \
          -library ip  \
          -version 4.* \
          -module_name xxv_ethernet_x4_0 \
          -dir ./ip_generation

set_property -dict [list CONFIG.LINE_RATE {10} \
                         CONFIG.NUM_OF_CORES {4} \
                         CONFIG.BASE_R_KR {BASE-R} \
                         CONFIG.INCLUDE_AXI4_INTERFACE {0} \
                         CONFIG.ENABLE_PIPELINE_REG {1}] \
             [get_ips xxv_ethernet_x4_0]

    set_property -dict [list CONFIG.GT_REF_CLK_FREQ {161.1328125}] \
                 [get_ips xxv_ethernet_x4_0]
~~~

3. use following connect directives in the connectivity_x4.cfg file:
~~~
[connectivity]
nk=ethernet_krnl_axis_x4:1:eth0
nk=data_fifo_krnl:4:df0.df1.df2.df3

stream_connect=eth0.rx0_axis:df0.rx_axis
stream_connect=eth0.rx1_axis:df1.rx_axis
stream_connect=eth0.rx2_axis:df2.rx_axis
stream_connect=eth0.rx3_axis:df3.rx_axis
stream_connect=df0.tx_axis:eth0.tx0_axis
stream_connect=df1.tx_axis:eth0.tx1_axis
stream_connect=df2.tx_axis:eth0.tx2_axis
stream_connect=df3.tx_axis:eth0.tx3_axis

connect=eth0/clk_gt_freerun:ii_level0_wire/ulp_m_aclk_freerun_ref_00
connect=io_clk_qsfp0_refclka_00:eth0/gt_refclk
connect=eth0/gt_port:io_gt_qsfp0_00
~~~

4. BTW, some routing issues are reported under 2022.1 version of Vitis toolchain for some certain board and config. If possible, please use 2022.2 version Vitis.

Thanks,

Gang

roger90810, 2022-12-07:
Hi, @Rampagee:

I change to Vitis 2022.2 and follow your instruction, now I can build it.

Thanks,

Roger

closed 2022-12-07 by roger90810