Xilinx/Vitis-Tutorials #337: vitis kernel hanging at cl:finish();/q.finish() on hardware.

nikhil2k10, 2022-12-16:
Hi,

I am using iwave's zu19eg development kit, I was trying to implement the vitis based designs on hardware. I started with basic vector_add code. When i am running the code on the hardware it is hanging at cl:finish();/q.finish().

Can anyone help me to resolve the issue?

@imrickysu @sgrace-xlnx  @dharapxilinx @xlinx-dachang

imrickysu, 2022-12-16:
Hi @nikhil2k10 , I assume you need to create a Vitis platform for your development kit. Some issue may occur during your customization. 

Hanging at cl:finish();/q.finish() may relate to the interrupt signal issue, or device tree for interrupt mapping. 

Since this issue is not directly caused by tutorials, I'll close this issue and suggest you ask this question on Xilinx forums. https://forums.xilinx.com/ and share the link here.
There are more people supporting the Xilinx forums. We can also follow up on the forums.

nikhil2k10, 2022-12-16:
Hi @imrickysu ,

Thanks for you comments. will check the interrupts signals, if there is any issue while building the device tree.
Also , I have raised a question in the forums. I am sharing the link below.

https://support.xilinx.com/s/question/0D54U000061gSB2SAM/vitis-kernel-hanging-at-clfinish-on-hardware?language=en_US

nikhil2k10, 2022-12-20:
Hi @imrickysu ,

I am attaching my device tree file here, Can you please help me find the issue in device tree?
[system_user_dtsi.txt](https://github.com/Xilinx/Vitis-Tutorials/files/10265439/system_user_dtsi.txt)

imrickysu, 2022-12-21:
Hi @nikhil2k10 , line 280 `interrupt-parent = <&axi_intc_0>;` is commented out. It's needed for zyxclmm_drm. You need to find out the interrupt controller node name and replace "axi_intc_0".

AnusheelXilinx, 2024-12-21:
Closing the issue as there are no open queries. 

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx