Xilinx/Vitis-Tutorials #339: FIR AIE tutorial hangs while running on rne_emu stage

praveen0447, 2022-12-17:
Hi,

I am trying the 2022.2 version of AIE FIR tutorial on an emulator. I was able to create the application and xclbin with the steps mentioned in the [link](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Design_Tutorials/07-firFilter_AIEvsHLS/AIE). However, when I launch the application it hangs at 
_"A72-Info: (Step 5) Wait for Data Mover Kernels to Complete  "_

Please let me know what am I missing here. Attached is the log for your reference.[log.txt](https://github.com/Xilinx/Vitis-Tutorials/files/10251920/log.txt)

The only point I deviated from the example is, that the VCK190 Vitis Platform for 2022.2 is not available. I used the one shipped with /tools/Xilinx/Vitis/2022.2/base_platforms/xilinx_vck190_base_202220_1 (Copied them to tools/Xilinx/Vitis/2022.2/platforms).

Thanks,
Praveen.

SURUTHI1605, 2022-12-20:
Hi @praveen0447 
In design is having xrtRunWait option - will Wait for a run to complete https://xilinx.github.io/XRT/master/html/xrt_native.main.html. It may take sometime to complete.Can you share me the TAP and FILTER SIZE in which you are experincing this issue?

SURUTHI1605, 2022-12-21:
Hi @praveen0447 ,
In 2022.1 version, this issue won't be there and in hw_emu the design will take max 10 minutes to complete. I'm in that debugging process and will let you know the solution once it's done.

praveen0447, 2022-12-23:
Hi @SURUTHI1605 , 

We tried @xflorentw suggested fix with no luck.  
Let me know if we can try the 2022.2 tutorial example on hardware. Do you foresee any similar issues with the hardware? 
Do you want us to proceed with running the tutorial  2022.1 version for the emulator/hardware until there is a fix for afore mentioned issue? 
  
I used the default sizes mentioned in the examples.I assume the filter size is 1 and the taps are 15, the window size is 256. Below generated build folder
build/fir_15_taps/x1_firs/winSz_256/

Thanks, 
Praveen.

SURUTHI1605, 2022-12-23:
Hi @praveen0447 
The issue is with some handshaking signals.We found a fix and about to checked-in in repo .For now please use tutorials of 2022.1 version for emulator/hardware.

Thanks,
suruthi

praveen0447, 2023-01-04:
Hi @SURUTHI1605 

I tried to run the example using 2022.1. I am facing compilation error during make graph step. Attached is the log([fir_aie_2022_1_compilationError_make_graph.txt](https://github.com/Xilinx/Vitis-Tutorials/files/10343474/fir_aie_2022_1_compilationError_make_graph.txt)) for you reference  

Please let me know if I am missing anything here.

Thanks,
Praveen.

SURUTHI1605, 2023-01-04:
Can you share me the vitis build and platform details you have sourced?

xflorentw, 2023-01-04:
@praveen0447 
Make sure you are also using the version 2022.1 of the DSPlib (Vitis Libraries)

praveen0447, 2023-01-10:
Hi @SURUTHI1605 

I used the vck190 platform for 2022.1 and Vitis 2022.1 for which I got issues. I've attached the log in the previous reply as the paths which mention my build and toolchain details.

@xflorentw Even with the Vitis 2022.1 DSP library, I see the same issue.Since the compilation error is related to the graph class constructor does the DSP library matter here?

Thanks,
Praveen.

AnusheelXilinx, 2024-12-21:
Hi @praveen0447, are you still seeing this issue with the latest 2024.2 release?

Thanks
Anusheel

closed 2025-01-10 by AnusheelXilinx