Xilinx/Vitis-Tutorials #342: Failed to fetch URL git://gitenterprise.xilinx.com/aisw/unilog.git;protocol=https;branch=dev, attempting MIRRORS if available

belanasaikiran, 2022-12-21:
WARNING: unilog-1.4.0-r0 do_fetch: Failed to fetch URL git://[gitenterprise.xilinx.com/aisw/unilo](https://gitenterprise.xilinx.com/aisw/unilo) g.git;protocol=https;branch=dev, attempting MIRRORS if available

The SState-Cache URLs seem to be missing from the last few days. Please update as it throws an error while building Petalinux Images.

Version used: 2021.1

I think the URLs have been removed.

Kindly, suggest new URLs and instructions to add SState-Cache verifications

imrickysu, 2022-12-21:
Hi @belanasaikiran , may I know in which tutorial and step this issue occurs?

belanasaikiran, 2022-12-23:
I'm using `petalinux-build` command from the below documentation link:

[https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step2.md#build-petalinux-images:~:text=Build%20PetaLinux%20Images](https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step2.md#build-petalinux-images:~:text=Build%20PetaLinux%20Imagesl)

The `sstate-cache` URLs seem to be expired a few days ago.

imrickysu, 2023-01-13:
Hi @belanasaikiran , I find this error message is pointing to Vitis-AI layer. May I know which tutorial are you working on? We need to do further investigation. Thanks.

belanasaikiran, 2023-01-17:
I'm using `petalinux 2021.1` version to build my Image. The tutorial is made for `ZCU104` as an example and the `state-cache` will be valid for all the MPSoC boards.

[Tutorial Link](https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step2.md#build-petalinux-images)

Please let us know once the URLs are updated or do we need to add any other URLs to `yocto-layers`

AnusheelXilinx, 2024-12-21:
Hi @sven2314, can you check if the above concern is addressed in the latest 2024.2 release?

Thanks
Anusheel