Xilinx/Vitis-Tutorials #343: Issue in running tutorial (01-rtl_kernel_workflow)

MahmoudElShimy96, 2022-12-23:
I build the source files for this tutorial (01-rtl_kernel_workflow) for board alveo u280 and when i run i get this issue:
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted

I tried to reset the board as mentioned here: https://support.xilinx.com/s/question/0D52E00006yn3ZpSAI/i-get-an-stdbadalloc-error-when-allocating-a-buffer-in-global-memory-for-the-example-in-01rtlkernelworkflow?language=en_US

But i get the same issue and i validated the board and it is working fine.
I am using Vitis 2022.1 and xrt 2022.1

randyh62, 2022-12-29:
Have you been able to get any other tutorials to run on your U280 card? Or any of the [Vitis_Accel_Examples](https://github.com/Xilinx/Vitis_Accel_Examples/tree/2022.1)?  Also, as this is not a specific issue related to the tutorial, I would suggest you post the issue on the support forum.

imrickysu, 2023-01-13:
Hi @MahmoudElShimy96 , `std::bac_alloc` is a common XRT issue. It can be caused by multiple reasons. For example, if XRT cannot find your xclbin file, it will report this issue.

Since you changed the design environment, we are not sure whether this issue is introduced by your modification, your hardware environment, or the tutorial design. Like @randyh62  mentioned, please make cross check with Vitis_Accel_Examples. If other designs can work, but only this tutorial doesn't work in your environment, please let us know.

For now, I'm closing this issue due to long time no response. Please reopen this issue if you have new findings.

closed 2023-01-13 by imrickysu