Xilinx/Vitis-Tutorials #344: How to benchmark AI Engine code on hardware

nhkrishna, 2022-12-23:
Hello Everyone,
I would like to benchmark the AI Engine code on the hardware. Can anyone let me know, any XILINX tool that used for timing the CODE that running on the AI Engine only on the hardware (VCK190).
I would like to insert the timing code in the LeNet tutorial.

imrickysu, 2023-01-03:
Hi @nhkrishna , please check whether the following articles are helpful:

- https://docs.xilinx.com/r/en-US/ug1076-ai-engine-environment/Profiling-Graph-Latency
- https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/09-debug-walkthrough/Debug6_pr.md
- https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/09-debug-walkthrough/Debug4_et.md#Number-of-event-trace-streams-methodology

cc @JithinPillai

Hari-MathWorks, 2023-01-09:
Hi,

I tried to follow [13-aie-performance-analysis](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Feature_Tutorials/13-aie-performance-analysis), It was using event handlers to profile the **<graphObj>.run()**. But, In my case, I'm trying to
to use profiling code in the [01-aie_lenet_tutorial](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial) where it will not use **<graphObj>.run()** instead it will use xrtGraph... API's. I couldn't find any example that profiles xrtGraph API's. 
Can you please let me know how we can get the profiling results in tutorial examples where we use the xrtGrap API's in host file.

Thanks,
Hari

SURUTHI1605, 2023-02-09:
Which Vitis build you are using(2022.2/2023.1)?

AnusheelXilinx, 2024-12-21:
Closing the issue as there were no further updates.

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx