Xilinx/Vitis-Tutorials #348: LeNet AIE tutorial hangs with EN_TRACE=1 while running on VCK190 hardware

Hari-MathWorks, 2023-01-02:
Hello Everyone,

I'm trying out the 2022.2 version [01-aie_lenet_tutorial](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial). I was able to build the example and successfully deployed on to the hardware. Thanks for the detailed steps in the tutorial page.

However, When I tried to see the throughput measurement details by building the application using the "EN_TRACE=1" option mentioned in the tutorial [link](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial#throughput-measurement-details), application hangs at the runtime.

Please see the below screenshot for the Tera Term command window output.
[image]

Is this behavior expected? Let me know is there anyway to bypass this behavior and see the throughput details.

Thanks,
Hari

SURUTHI1605, 2023-01-03:
Can you share the vitis version , build name and run commands you had used?

nhkrishna, 2023-01-03:
Below are version and commands details:
Vitis Version: 2022.2

[01-aie_lenet_tutorial](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial) build steps used:
1. make kernels TARGET=hw EN_TRACE=1
2. make graph TARGET=hw EN_TRACE=1
3. make xsa TARGET=hw EN_TRACE=1
4. make application TARGET=hw EN_TRACE=1
5. make package TARGET=hw EN_TRACE=1

Runtime steps:
1. Followed the same steps mentioned in the tutorial page. It seems application hanged while running the "dma hls"

Let me know if you need any information.

Hari-MathWorks, 2023-01-03:
Below are version and commands details:
Vitis Version: 2022.2

[01-aie_lenet_tutorial](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial) build steps used:

make kernels TARGET=hw EN_TRACE=1
make graph TARGET=hw EN_TRACE=1
make xsa TARGET=hw EN_TRACE=1
make application TARGET=hw EN_TRACE=1
make package TARGET=hw EN_TRACE=1
Runtime steps:

Followed the same steps mentioned in the tutorial page. It seems application hanged while running the "dma hls"

Basically, I would like to see the benchmark results for the AI Engine. Let me know if you need any information.

SURUTHI1605, 2023-01-03:
Can you share the result of "which vivado" or "which vitis" after sourcing the vitis 2022.2 build?

Hari-MathWorks, 2023-01-03:
Output of "which vitis":
`$(XILINX_VITIS_INSTALL_DIR)/Vitis/2022.2/bin/vitis`

Output of `platforminfo --list | grep -m 1 -A 9 vck190_base`
[image]

SURUTHI1605, 2023-01-03:
Please share the folders in base_platforms repository.

SURUTHI1605, 2023-01-03:
You're using "xilinx_vck190_base_dfx_202220_1" platform to run this design. But it has been designed for "xilinx_vck190_base_202220_1" platform. Please use the below mentioned platform and try to run the design.

Confirm you have the VCK190 Production Base Platform.
"platforminfo --list | grep -m 1 -A 9 vck190_base"

Output of the above command should be as follows:

"baseName": "xilinx_vck190_base_202220_1",
   

Thanks

Hari-MathWorks, 2023-01-03:
Hi @SURUTHI1605 ,

Below are the folders presented in the base_platforms:
[image]

It seems, I already have "Xilinx_vck_190_base_202220_1" folder.  If I use the vck190_base_2022 then it will show correct base name.
Below is the output of `platforminfo --list | grep -m 1 -A 9 vck190_base_2022`
[image]

I also confirmed that in the compilation logs, platforms files are picked from the `Xilinx_vck_190_base_202220_1` folder.

Hari-MathWorks, 2023-01-03:
Also, I have modified the core003.cc implementation. It has optimized kernel code. I have modified to use the plain c code for matrix multiplication. I'm doing this exercise to see the performance improvement with the AIE kernel code. 

Please see the attached "core03.cc" file to the modifications I made.
[core03.cc.tar.gz](https://github.com/Xilinx/Vitis-Tutorials/files/10336847/core03.cc.tar.gz)

SURUTHI1605, 2023-01-04:
Hi @Hari-MathWorks , from your side have you done any changes other than core03.cc file?
For the design we need to check first in hardware emulation platform and then in hardware. Since you've done some changes in AIE kernel we need to check and verify the handshaking signal and output of AIE  in emulation platform. I can try this from my side.

Thanks

Hari-MathWorks, 2023-01-04:
Hi @SURUTHI1605 ,

If I run the [01-aie_lenet_tutorial](https://github.com/Xilinx/Vitis-Tutorials/tree/2022.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial) without any modifications then it worked fine. I don't see any issues. But, If i use modified(Specifically used plain "C" code for matrix multiplications) the core03.cc mentioned in the above comment then, I was seeing the crash. I'm also looking into this. Why this was happening?

I have also experimented by removing matrix multiplication aie code in core03.cc then application running with golden output mismatch error at the end. But, this is expected.

SURUTHI1605, 2023-02-09:
Hi @Hari-MathWorks ,
Are you able to build the design?

AnusheelXilinx, 2024-12-21:
Closing the issue as there are no further open queries.

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx