Xilinx/Vitis-Tutorials #360: 13-aie-performance-analysis: Facing runtime errors when I try to find the graph latency using 13-aie-performance-analysis

Hari-MathWorks, 2023-01-19:
**Describe the issue**
I was able to successfully run the "**13-aie-performance-analysis**" on both Hardware emulation and actual hardware with out any modifications. But If I use the `event::start_profiling(gr.in, gr.dataout, event::io_stream_start_difference_cycles)` to find the graph latency, then I was facing the issues.
Below is the screenshot of the modifications I made to the [host.cpp](https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/13-aie-performance-analysis/testcase_ssfifo/sw/host.cpp) file:
[image]

You can find the complete modified host.cpp file from below:
[host_modified.cpp.txt](https://github.com/Xilinx/Vitis-Tutorials/files/10456764/host_modified.cpp.txt)

**To Reproduce**
1. Modify the `Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/13-aie-performance-analysis/testcase_ssfifo/sw/host.cpp` file with modifications mentioned in above section.
2. Run the hardware emulation step mentioned in this [link](https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Feature_Tutorials/13-aie-performance-analysis/aie_execution_measurement.md) i.e., `make run_hw_emu`.
3. In the emulator run `host.exe a.xclbin`
4. You will see below error:
[image]

**Your environment: **
- OS: Ubuntu 22.04.1 LTS
- Software Version: 2022.2
- Hardware (if applicable): VCK190

xflorentw, 2023-02-06:
This question has been asked on the forums and technical support is investigating the issue:
https://support.xilinx.com/s/question/0D54U00006FZmseSAD/facing-runtime-errors-when-i-try-to-find-the-graph-latency-using-13aieperformanceanalysis-tutorial

AnusheelXilinx, 2024-12-21:
Issue is fixed as per the forums discussion, closing the issue. 

Thanks
Anusheel

closed 2024-12-21 by AnusheelXilinx