Xilinx/Vitis-Tutorials #361: Cannot integrate the DPU Kernel (RTL Kernel) on a custom design

davendramaharaj1, 2023-02-08:
Cannot select the DPU Kernel (RTL Kernel) as a template
I am having an issue integrating the DPU into my Vitis project. I am following the procedure outlined in the README under DPUCZDX8G/prj/Vitis for Vitis 2.5. I extracted the DPUCZDX8G.tar.,gz file which created the DPUCZDX8G folder. Then I added it to Library Repositories as follows:

[image]

I tried creating an application project, selected my custom platform that I created (kv260_custom.xpfm). Then I set the following:

- sysroot path = <kv260_custom_pkg>/sysroot/cortexa72-cortexa53-xilinx-linux,
- rootfs path = kv260_custom_petalinux/xilinx-kv260-starterkit-2022.2/images/linux/rootfs.ext4
- kernel image path = kv260_custom_plnx/xilinx-kv260-starterkit-2022.2/images/linux/Image.

After this, I tried to select "DPU Kernel (RTL Kernel) and I only see a blank screen, no information. Moreover, I cannot move on as "Next" does not work as shown here:

[image]

 
I am not sure what's supposed to happen here. The following are the system configurations:

- OS: Ubuntu 22.04
- Vitis: 2022.2
- Board: KV260
- DPU TRD from Vitis AI 2.5

xflorentw, 2023-02-08:
Hi @davendramaharaj1 

This issue section is for issues related to tutorials which are part of this current Vitis-Tutorials repository.
You question is not about an issue of any of the tutorials present in this repository.

For general questions on the Vitis tools, please ask on the support community forums: https://support.xilinx.com/s/

imrickysu, 2023-02-09:
Hi @sven2314 , if you have related experience, please share with @davendramaharaj1 .

Hi @davendramaharaj1 , checking with the community forums is recommended. We'll share our experiences when possible.

davendramaharaj1, 2023-02-09:
Hi @imrickysu, I have also posted on the Xilinx forums under Vitis AI and Vitis Acceleration & Acceleration here: https://support.xilinx.com/s/question/0D54U00006MlYUnSAN/cannot-select-the-dpu-kernel-rtl-kernel-as-a-template?language=en_US

but no one has answered offered advice yet.

sven2314, 2023-02-10:
@davendramaharaj1 
Hi,  I tried this method to add DPU kernel. When I create an application it can detect the DPU application.
[image]
Just click finish it can complete the project creation.
Vitis 2022.2
Ubuntu
VAI 2.5

davendramaharaj1, 2023-02-10:
@sven2314 If you look at my first screenshot, I have already added the path to the DPU library I downloaded for Vitis 2.5. Then, I tried to repeat the steps you just posted after creating an application. If you look at the 2nd screenshot, I tried to add the DPU Kernel and the box to the right is blank. Also, I could not move on by clicking "Next".

davendramaharaj1, 2023-02-11:
I see got it. For some reason, my vitis gui does not generate any info on the right text box so it's a bit weird. But I got the project creation working. Thanks anyway.

closed 2023-03-30 by davendramaharaj1